flyte/mqtt-io #224: Home Assistant MQTT discovery. Unique device not created

lancasterJ, 2021-05-02:
When mqtt-io is installed on more than one Pi and ha_discovery is enabled the configuration messages sent to MQTT Discovery are merged to create a single HA device.

This occurs because the device.identifies variable sent in the config payload is not unique. A list of two IDs is sent, the first is always “mqtt-io” and the second is the unique client_id.  HA merges the devices because they all have “mqtt-io” as one of their IDs.

To resolve the issue ensure the device IDs are unique by making the following change to home_assisyant.py.

At line 37
change 
`	identifiers=["mqtt-io", mqtt_options.client_id]`
to
`	identifiers=[mqtt_options.client_id]`

snowgumrd, 2022-01-06:
I had this same issue! Was mucking around with names for a half a day before I read this! I suspected it was that but had no idea how to find it. Thanks @lancasterJ !

andystewart999, 2022-01-06:
That's crazy!  I literally just discovered this issue about 6 hours ago... and this fix 5 minutes ago!  Thanks again @lancasterJ