flyte/mqtt-io #225: Can't get it up and running: No module named mqtt_io

Kimovitzh, 2021-05-07:
Hi

I'm not having any lock running MQTT-IO on my Raspberry.
when I execute the` python3 -m mqtt_io config.yml`
all I get is:
_/usr/local/bin/python3.9: No module named mqtt_io_

I then saw a post about running it as a supervisor but there I get:
_mqtt_io                          BACKOFF   Exited too quickly (process log may have details)_

I must be missing something.

BenjiU, 2021-05-07:
hi, may you please post your command, when you use sudo? When executing as sudo, you need to change the path to your config.yml, because with sudo a different location is used...

Kimovitzh, 2021-05-08:
- [ ] 

> hi, may you please post your command, when you use sudo? When executing as sudo, you need to change the path to your config.yml, because with sudo a different location is used...

Ehm, I just installed python, the did the `pip3 install mqtt-io` and entered `python3 -m mqtt_io config.yml`.

Looking for the config.yml ... I can't seem to locate it 🤔

Sorry for being such a noob, I'm just used to close to plug and play stuff like PiHole, zigbee2mqtt and Mosquitto.

BenjiU, 2021-05-17:
Sorry for the delay... so you don't have a `config.yml` file? You should create on, e.g. copying from the example file `config.example.yml`. But with your configuration.

flyte, 2021-05-17:
Have a look at the documentation which has some examples of config files https://mqtt-io.app/2.2.6/#/config/scenarios

This project is not really a plug and play solution because it can be used in infinite configurations with all sorts of different hardware. A config file must be created by you to suit your hardware.

closed 2021-06-30 by BenjiU