flyte/mqtt-io #228: Not able to get it running...

MrMennens, 2021-05-17:
Please consider joining the [Discord server](https://discord.gg/gWyV9W4) in order to get support. This is likely to be the quickest way for somebody to help you.

If you'd still like to add your support request as a GitHub Issue, please fill in the following template as best you can:

- [ ] I have read the [README.md](https://github.com/flyte/pi-mqtt-gpio/blob/develop/README.md) file and visited the [Documentation](https://mqtt-io.app/) to see if the information there helps.

**Action description**
python3 -m mqtt_io config.yml

**Previous attempts**
Sudo / wo sudo
reinstall pip packages

**Error messages and traceback**
```
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.7/dist-packages/mqtt_io/__main__.py", line 85, in <module>
    main()
  File "/usr/local/lib/python3.7/dist-packages/mqtt_io/__main__.py", line 52, in main
    config = load_main_config(args.config)
  File "/usr/local/lib/python3.7/dist-packages/mqtt_io/config/__init__.py", line 175, in load_main_config
    with open(path, "r") as stream:

```

**Config**
```yaml
mqtt:
  host: localhost
  topic_prefix: test

# GPIO
gpio_modules:
  # Use the Raspberry Pi built-in GPIO
  - name: rpi
    module: raspberrypi

#digital_inputs:
  # Pin 0 is an input connected to a doorbell button
#  - name: doorbell
#    module: rpi
#    pin: 0

#digital_outputs:
#  # Pin 1 is an output connected to a light
#  - name: porch_light
#    module: rpi
#    pin: 1

# Sensors
sensor_modules:
  # An LM75 sensor attached to the I2C bus
  - name: ina219_sensor
    module: ina219
    i2c_bus_num: 1
    chip_addr: 0x40

sensor_inputs:
  # The configuration of the specific sensor value to use (LM75 only has temperature)
  - name: teest
    module: ina219_sensor

# Streams
stream_modules:
  # A serial port to communicate with the house alarm system
  - name: alarm_system
    module: serial
    device: /dev/ttyUSB0
    baud: 9600

```

**Hardware**
 - Platform: Raspberry Pi
 - Connected hardware:  GPIO INA219

**System:**
 - OS: Raspbian
 - Python version: 
python --version
Python 2.7.16
python3 --version
Python 3.7.3

 - User you're running as: pi / sudo
 - Using a virtualenv?: no

**Additional context**
Not sure were to go from here...

flyte, 2021-05-17:
It looks like the traceback is missing the bit at the bottom that says what the actual issue is. Could you paste the whole lot please?

MrMennens, 2021-05-18:
damn...

typo 

config.yml has to be config.yaml  

sorry for opening an issue...

MrMennens, 2021-05-18:
> It looks like the traceback is missing the bit at the bottom that says what the actual issue is. Could you paste the whole lot please?

flyte, 2021-05-18:
Yeah, adding documentation/examples for each of the modules is definitely on the list.

I'm not sure where your error is coming from though. Would you mind adding the following to the top of your config file? It'll send the error, along with a bunch of details to sentry so I can see what's going on:

```yaml
reporting:
  enabled: yes
  issue_id: 228
```

Information about this feature is [here in the documentation](https://mqtt-io.app/2.2.6/#/config/reference/reporting/)

MrMennens, 2021-05-18:
Yeah, noticed your todo list... i'm a bit to early maybe ;)
if i can be of any help with my ina219, just let me know!

I added the code you requested, hope you can seee it in sentry! (tried 3 different versions of the config.yaml file)

Good luck

frabnet, 2021-12-08:
Hi, I have a similar problem:

config.yml
```
mqtt:
  host: iotsrv.lan
  topic_prefix: testmqtt

sensor_modules:
  - name: myina
    module: ina219
    i2c_bus_num: 0
    chip_addr: 0x40
    shunt_ohms: 0.1

sensor_inputs:
   - name: voltage_ina
     module: myina
```

frab@orangepione:~$ python3 -m mqtt_io config.yml 
```
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/frab/.local/lib/python3.7/site-packages/mqtt_io/__main__.py", line 85, in <module>
    main()
  File "/home/frab/.local/lib/python3.7/site-packages/mqtt_io/__main__.py", line 80, in main
    mqtt_gpio.run()
  File "/home/frab/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 1229, in run
    self._init_sensor_modules()
  File "/home/frab/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 256, in _init_sensor_modules
    sens_config, "sensor", self.config["options"]["install_requirements"]
  File "/home/frab/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 116, in _init_module
    "%s.%s.%s" % (MODULE_IMPORT_PATH, module_type, module_config["module"])
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'mqtt_io.modules.sensor.ina219'
```

Do you think it's the same problem?

lada2102, 2022-03-20:
Hi, I have a problem with mqtt-io on rp3b+/raspbian buster/Python3.7 but on bullseye everything works fine, and now i have errors, i think, that's a problem with connecting to mqtt server.
pi@raspberrypi:~ $ sudo python3 -m mqtt_io /home/pi/config.yml
2022-03-20 02:05:59 mqtt_io.server [INFO] Connecting to MQTT...
2022-03-20 02:05:59 mqtt_io.server [INFO] Digital input 'drzwi_kotlownia' value changed to True
2022-03-20 02:05:59 mqtt_io.server [INFO] Digital input 'okno_kotlownia' value changed to True
2022-03-20 02:05:59 mqtt_io.server [INFO] Digital input 'okno_kuchnia' value changed to True
2022-03-20 02:05:59 mqtt_io.server [INFO] Digital input 'okno_jadalnia' value changed to True
2022-03-20 02:05:59 mqtt_io.server [INFO] Digital input 'okno_salon' value changed to True
2022-03-20 02:05:59 mqtt_io.server [INFO] Digital input 'okno_gabinet' value changed to True
2022-03-20 02:05:59 mqtt_io.server [INFO] Digital input 'ruch_garaz' value changed to True
2022-03-20 02:05:59 mqtt_io.server [INFO] Digital input 'ruch_kotlownia' value changed to True
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/asyncio_mqtt/client.py", line 79, in connect
    await loop.run_in_executor(None, self._client.connect, self._hostname, self._port, 60)
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.7/dist-packages/paho/mqtt/client.py", line 914, in connect
    return self.reconnect()
  File "/usr/local/lib/python3.7/dist-packages/paho/mqtt/client.py", line 1044, in reconnect
    sock = self._create_socket_connection()
  File "/usr/local/lib/python3.7/dist-packages/paho/mqtt/client.py", line 3685, in _create_socket_connection
    return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)
  File "/usr/lib/python3.7/socket.py", line 707, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
  File "/usr/lib/python3.7/socket.py", line 748, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/mqtt_io/mqtt/asyncio_mqtt.py", line 32, in inner
    await func(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/mqtt_io/mqtt/asyncio_mqtt.py", line 79, in connect
    await self._client.connect(timeout=timeout)
  File "/usr/local/lib/python3.7/dist-packages/asyncio_mqtt/client.py", line 88, in connect
    raise MqttError(str(error))
asyncio_mqtt.error.MqttError: [Errno -2] Name or service not known

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/mqtt_io/server.py", line 1158, in _main_loop
    await self._connect_mqtt()
  File "/usr/local/lib/python3.7/dist-packages/mqtt_io/server.py", line 585, in _connect_mqtt
    await self.mqtt.connect()
  File "/usr/local/lib/python3.7/dist-packages/mqtt_io/mqtt/asyncio_mqtt.py", line 34, in inner
    raise MQTTException from exc
mqtt_io.mqtt.MQTTException

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.7/dist-packages/mqtt_io/__main__.py", line 85, in <module>
    main()
  File "/usr/local/lib/python3.7/dist-packages/mqtt_io/__main__.py", line 80, in main
    mqtt_gpio.run()
  File "/usr/local/lib/python3.7/dist-packages/mqtt_io/server.py", line 1237, in run
    self.loop.run_until_complete(self._main_task)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "/usr/local/lib/python3.7/dist-packages/mqtt_io/server.py", line 1183, in _main_loop
    if reconnects_remaining is not None:
UnboundLocalError: local variable 'reconnects_remaining' referenced before assignment

config.yml
mqtt:
  host: 192.168.192.5
  user: "qmax"
  password: "hasspass123!"
  topic_prefix: alarm

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_inputs:
  - name: drzwi_kotlownia
    module: raspberrypi
    pin: 4
    on_payload: "open"
    off_payload: "closed"
    pullup: yes
    pulldown: no

digital_outputs:

sensor_modules:

sensor_inputs:

stream_modules:

Thank's.

BenjiU, 2022-03-20:
I don't know, what's the original problem of lada2102 is.

But we need to switch these both lines in the try catch statement, because reconnects_remaining is used in catch, when _connect_mqtt() fails:
                await self._connect_mqtt()
                reconnects_remaining = self.config["mqtt"]["reconnect_count"]

BenjiU, 2022-03-21:
@lada2102: where is host: 192.168.192.5 pointing to? Is it also on your raspberry or is it an external server for both of your scenarios: buster or bullseye?

lada2102, 2022-03-21:
@BenjiU : Host 192.168.192.5 is in Zerotier-one external network and i think that's the problem. In local network everything works fine. Currently i have Raspbian Bullseye. I try solve problem with connection and check mqqt-io again.

closed 2022-11-04 by BenjiU