flyte/mqtt-io #231: MQTT Actions not triggering even though subscriber messages are being seen

HumanSkunk, 2021-06-08:
- [X] I have read the [README.md](https://github.com/flyte/pi-mqtt-gpio/blob/develop/README.md) file and visited the [Documentation](https://mqtt-io.app/) to see if the information there helps.

**Describe the bug**

I have mqtt-io installed and managed through supervisor. I have found after an unknown amount of time mqtt-io fails to take action when an mqtt message is sent. I can see the messages being received on the MQTT broker and I can connect and see those messages from the Pi running mqtt-io however the GPIO pins are not being triggered. If I ```sudo supervisorctl restart mqtt-io``` then the actions will work perfectly. One solution is to run a cron job every night to restart the service but this is a clunky solution. I am unsure what the cause is and I dont know why restarting the supervisor service kicks it back into life. Running ```sudo supervisorctl status``` shows that the process is running without error and has had an uptime of since I last restarted the process.

**Expected behavior**

When sending the MQTT publish command I expect the action governed by the .yml file to take action. Restarting the Supervisor service makes the program behave as expected.

**Error messages and traceback**
```2021-05-31 18:20:01,477 INFO waiting for mqtt_io to stop
2021-05-31 18:20:01,833 INFO stopped: mqtt_io (exit status 1)
2021-05-31 18:20:01,848 INFO spawned: 'mqtt_io' with pid 4538
2021-05-31 18:20:02,851 INFO success: mqtt_io entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
```
These are the only logger entries that appear under the supervisor logs. Nothing indicating there is a fault.

**Config**
```mqtt:
  host: A.B.C.D
  topic_prefix: home
  user: pi
  password: <password>

# GPIO
gpio_modules:
  # Use the Raspberry Pi built-in GPIO
  - name: rpi
    module: raspberrypi

digital_outputs:
  - name: modem
    module: rpi
    pin: 17
  - name: router
    module: rpi
    pin: 4
```

**Hardware**
 - Platform: Raspberry Pi 3 A+
 - Connected hardware:  2 5V Relays triggering off GPIO pins 4 and 17

**System:**
 - OS: [e.g. Raspbian]
 - Python version: python3 --version Python 3.7.3
 - User you're running as: pi
 - Using a virtualenv?: yes

**Additional context**
Add any other context about the problem here.

flyte, 2021-06-08:
Yeah, this is a known issue at the moment. I've made quite a big update recently that's just waiting for me to have a bit more time to finish fixing the tests for, then I'll release a much more robust (hopefully!) version.

HumanSkunk, 2021-06-08:
Ah thats good to hear. Will work with clunky workaround for now then. I will keep an eye out for the next release or if there is anything you want me to test in the meantime let me know. I spent so long trying to get something working before I realised your program did exactly what I needed. Thanks for the effort you have put in so far.

flyte, 2021-06-08:
You can try the `feature/trio` branch if you're able to get it installed with [poetry](https://github.com/python-poetry/poetry). I don't currently have any 'developing' docs, but it should just be `poetry install` and then `poetry run python -m mqtt_io config.yml` off the top of my head.

flyte, 2021-06-08:
Alternatively, you can use a docker image (that's what I'm doing). I'm periodically adding new tags when I update my own stuff, so you can use whatever the newest tag is if you'd like https://hub.docker.com/r/flyte/mqtt-io/tags?page=1&ordering=last_updated

MTJoker, 2021-07-20:
Can it be that I face the same issue "the other way round"?
I have a push button connected to a Piface, and when I press it I normally see the corresponding MQTT message. After some time, it just stops working, and I don't see the MQTT message any more. If I remember right, also I don't see the log message "Digital input value changed" any more. If I `sudo supervisorctl restart mqtt-io`, everything starts working instantly again.
Is this the same issue or should I open a new one?

BenjiU, 2021-07-22:
Hi @MTJoker, I can't see your issue on my installation, that is running for weeks. About what periods are you talking? Hours, Days, Weeks?
Could it be a hardware issue? Maybe something with resistors and capacitors? When mqtt-io is restarted, maybe something on the pin configuration is changed via GPIO module (pullup/pulldown), that unloads e.g. the capacitor?

MTJoker, 2021-07-23:
Hi @BenjiU,
> About what periods are you talking? Hours, Days, Weeks?

It is days for sure, but I can't tell exactly (the button is my doorbell, so I don't notice immediately when it stops working... Currently it is running since 3 days). 

> Could it be a hardware issue? Maybe something with resistors and capacitors? When mqtt-io is restarted, maybe something on the pin configuration is changed via GPIO module (pullup/pulldown), that unloads e.g. the capacitor?

Could be, but I am quite sure that the same setup used to work in older versions, so I doubt it.
I am not using the GPIO pins of the raspi directly, but my button is connected to a PiFace. The wire to the button is rather long (approx. 10m).  It is a .8mm cable, normally the length shouldn't be an issue for this application.

BenjiU, 2021-07-23:
Hmm, than you should more often use food delivery: you would faster notice, when the doorbell does not work :-D 
Maybe it's an issue with the piface module? Do you have outputs on your piface? Or may you add one? If you recognice, that the input (doorbell) does not work any longer, you may try to trigger the output via mqtt. Maybe that gives a better hint, if piface is still working...
Or, if you are firm with that, you could add a _LOG in digital_input_poller(), and check, if the traces are still coming, when your doorbell does not work, or if the task crashed...
I had that with my i2c temp sensor: when there was a problem reading the sensor, the old version (pi_gpio_mqtt?) crashed. In the new version (mqtt_io), I haven't noticed that issue yet.

andystewart999, 2022-01-06:
@flyte - another data point for you.

I'd started experiencing this issue when running MQTT-IO via Supervisor (along with another python3 script, system_sensors).  For testing I disabled Supervisor entirely and instead ran MQTT-IO via an @reboot  cron job... all good.

The weird thing is, when re-enabling the Supervisor service and having it ONLY run the other python3 script, MQTT-IO started to fail to get messages again.  When I disabled Supervisor and rebooted it was all fine again.

I've only just reached this point so I may try and do some more testing but for now I'm going to stick with cron as the activator for both MQTT-IO and System_Sensors, although I'm sad to leave Supervisor behind for the moment.

I hope this helps,
Andy

andystewart999, 2022-01-13:
> @flyte - another data point for you.
>
> I'd started experiencing this issue when running MQTT-IO via Supervisor (along with another python3 script, system_sensors). For testing I disabled Supervisor [...]

Hmm - it was actually nothing to do with Supervisor (probably).  I started running [System_Sensors](https://github.com/Sennevds/system_sensors) and as soon as it runs it kills your app, or at least connectivity to it.  Time for more digging.