flyte/mqtt-io #232: Logging to file does not work in 2.2.6, works ok in 0.5.6

mihadoo, 2021-06-20:
When I add the following for logging to file then the file is not written in 2.2.6. In 0.5.6 this was working correctly:

```
logging:
  version: 1
  formatters:
    simple:
      format: "%(asctime)s %(name)s (%(levelname)s): %(message)s"
  handlers:
    console:
      class: logging.StreamHandler
      level: DEBUG
      formatter: simple
      stream: ext://sys.stdout
    file:
      class : logging.handlers.RotatingFileHandler
      level: DEBUG
      formatter: simple
      filename: /home/pi/log_pi_mqtt_gpio.log
      maxBytes: 100000
      backupCount: 3
  loggers:
    mqtt_gpio:
      level: INFO
      handlers: [console,file]
      propagate: yes 
```

Any idea what could be wrong with this?

BenjiU, 2021-06-30:
Hi, sorry for the delay. What exactly is not working? Just writing to /home/pi/log_pi_mqtt_gpio.log? But DEBUG info is written to the console?
I'll try to reproduce it. What is your execution command? [sudo] python3 -m mqtt_io config.yml

mihadoo, 2021-06-30:
Hi

The log file is not written into /home/pi

The execution command is: 

`sudo python3 -m mqtt_io /home/pi/pi_mqtt_gpio.yml`

The complete pi_mqtt_gpio.yml file is:

```
mqtt:
  host: 192.168.208.215
  port: 1883
  user: "openhabian"
  password: "openhabianpw"
  topic_prefix: bridge/t15/raspi5/gpio

logging:
  version: 1
  formatters:
    simple:
      format: "%(asctime)s %(name)s (%(levelname)s): %(message)s"
  handlers:
    console:
      class: logging.StreamHandler
      level: DEBUG
      formatter: simple
      stream: ext://sys.stdout
    file:
      class : logging.handlers.RotatingFileHandler
      level: DEBUG
      formatter: simple
      filename: /home/pi/log_pi_mqtt_gpio.log
      maxBytes: 100000
      backupCount: 3
  loggers:
    mqtt_gpio:
      level: INFO
      handlers: [console,file]
      propagate: yes

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_outputs:
  - name: garagelight
    module: raspberrypi
    pin: 26
    on_payload: "ON"
    off_payload: "OFF"
    initial: high  # This optional value controls the initial state of the pin before receipt of any messages from MQTT. Valid options are 'low' and 'high'.
    inverted: yes  # This pin may control an open-collector output which is "on" when the output is "low".
    retain: yes # This option value controls if the message is retained. Default is no.

  - name: garageventilator
    module: raspberrypi
    pin: 19
    on_payload: "ON"
    off_payload: "OFF"
    initial: high  # This optional value controls the initial state of the pin before receipt of any messages from MQTT. Valid options are 'low' and 'high'.
    inverted: yes  # This pin may control an open-collector output which is "on" when the output is "low".
    retain: yes # This option value controls if the message is retained. Default is no.

  - name: garagedoor
    module: raspberrypi
    pin: 5
    on_payload: "ON"
    off_payload: "OFF"
    initial: high  # This optional value controls the initial state of the pin before receipt of any messages from MQTT. Valid options are 'low' and 'high'.
    inverted: yes  # This pin may control an open-collector output which is "on" when the output is "low".
    retain: yes # This option value controls if the message is retained. Default is no.

  - name: gardendoor
    module: raspberrypi
    pin: 6
    on_payload: "ON"
    off_payload: "OFF"
    initial: high  # This optional value controls the initial state of the pin before receipt of any messages from MQTT. Valid options are 'low' and 'high'.
    inverted: yes  # This pin may control an open-collector output which is "on" when the output is "low".
    retain: yes # This option value controls if the message is retained. Default is no.

  - name: gardendoorright
    module: raspberrypi
    pin: 24
    on_payload: "ON"
    off_payload: "OFF"
    initial: low  # This optional value controls the initial state of the pin before receipt of any messages from MQTT. Valid options are 'low' and 'high'.
    inverted: no  # This pin may control an open-collector output which is "on" when the output is "low".
    retain: yes # This option value controls if the message is retained. Default is no.

  - name: garagetempsensorvoltage
    module: raspberrypi
    pin: 13
    on_payload: "ON"
    off_payload: "OFF"
    initial: high  # This optional value controls the initial state of the pin before receipt of any messages from MQTT. Valid options are 'low' and 'high'.
    inverted: yes  # This pin may control an open-collector output which is "on" when the output is "low".
    retain: yes # This option value controls if the message is retained. Default is no.

sensor_modules:
  - name: dht22_outside
    module: dht22
    type: DHT22 # can be  DHT11, DHT22 or AM2302
    pin: 15

  - name: dht22_inside
    module: dht22
    type: DHT22 # can be  DHT11, DHT22 or AM2302
    pin: 4

sensor_inputs:
  - name: garageoutsidetemperaturet
    module: dht22_outside
    interval: 60 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 4 # number of digits to be round
    type: temperature # Can be temperature or humidity
      
  - name: garageoutsidehumidity 
    module: dht22_outside
    interval: 60 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 4 # number of digits to be round
    type: humidity # Can be temperature or humidity   

  - name: garageinsidetemperaturet
    module: dht22_inside
    interval: 60 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 4 # number of digits to be round
    type: temperature # Can be temperature or humidity
      
  - name: garageinsidehumidity 
    module: dht22_inside
    interval: 60 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 4 # number of digits to be round
    type: humidity # Can be temperature or humidity   

digital_inputs:
  - name: garagedoorsensor
    module: raspberrypi
    pin: 22
    on_payload: "ON"
    off_payload: "OFF"
    pullup: no
    pulldown: yes

  - name: gardendoorsensor
    module: raspberrypi
    pin: 25
    on_payload: "ON"
    off_payload: "OFF"
    pullup: no
    pulldown: yes

  - name: gardenlightsensor
    module: raspberrypi
    pin: 21
    on_payload: "ON"
    off_payload: "OFF"
    pullup: no
    pulldown: yes

  - name: gardenmotionsensor
    module: raspberrypi
    pin: 17
    on_payload: "OFF"
    off_payload: "ON"
    pullup: no
    pulldown: yes 

```

flyte, 2021-06-30:
Ah, the issue is that the 'logger' should be called 'mqtt_io' now. It's
because of the large restructuring and name change.

Have a look at the default setting in
https://mqtt-io.app/2.2.6/#/config/reference/logging/

Sorry that it's formatted so badly in the example! I'll have to sort that
out at some point.

mihadoo, 2021-07-13:
Thank you, this worked. I am posting the complete file so that anyone else interested sees a working config file:

```
mqtt:
  host: 192.168.218.21
  port: 1883
  user: "openhabian"
  password: "openhabianpw"
  topic_prefix: bridge/t15/raspi5/gpio

logging:
  version: 1
  formatters:
    simple:
      format: "%(asctime)s %(name)s (%(levelname)s): %(message)s"
  handlers:
    console:
      class: logging.StreamHandler
      level: DEBUG
      formatter: simple
      stream: ext://sys.stdout
    file:
      class : logging.handlers.RotatingFileHandler
      level: DEBUG
      formatter: simple
      filename: /home/pi/log_pi_mqtt_gpio.log
      maxBytes: 100000
      backupCount: 3
  loggers:
    mqtt_io:
      level: INFO
      handlers: [console,file]
      propagate: yes

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_outputs:
  - name: garagelight
    module: raspberrypi
    pin: 26
    on_payload: "ON"
    off_payload: "OFF"
    initial: high  # This optional value controls the initial state of the pin before receipt of any messages from MQTT. Valid options are 'low' and 'high'.
    inverted: yes  # This pin may control an open-collector output which is "on" when the output is "low".
    retain: yes # This option value controls if the message is retained. Default is no.

  - name: garageventilator
    module: raspberrypi
    pin: 19
    on_payload: "ON"
    off_payload: "OFF"
    initial: high  # This optional value controls the initial state of the pin before receipt of any messages from MQTT. Valid options are 'low' and 'high'.
    inverted: yes  # This pin may control an open-collector output which is "on" when the output is "low".
    retain: yes # This option value controls if the message is retained. Default is no.

  - name: garagedoor
    module: raspberrypi
    pin: 5
    on_payload: "ON"
    off_payload: "OFF"
    initial: high  # This optional value controls the initial state of the pin before receipt of any messages from MQTT. Valid options are 'low' and 'high'.
    inverted: yes  # This pin may control an open-collector output which is "on" when the output is "low".
    retain: yes # This option value controls if the message is retained. Default is no.

  - name: gardendoor
    module: raspberrypi
    pin: 6
    on_payload: "ON"
    off_payload: "OFF"
    initial: high  # This optional value controls the initial state of the pin before receipt of any messages from MQTT. Valid options are 'low' and 'high'.
    inverted: yes  # This pin may control an open-collector output which is "on" when the output is "low".
    retain: yes # This option value controls if the message is retained. Default is no.

  - name: gardendoorright
    module: raspberrypi
    pin: 24
    on_payload: "ON"
    off_payload: "OFF"
    initial: low  # This optional value controls the initial state of the pin before receipt of any messages from MQTT. Valid options are 'low' and 'high'.
    inverted: no  # This pin may control an open-collector output which is "on" when the output is "low".
    retain: yes # This option value controls if the message is retained. Default is no.

  - name: garagetempsensorvoltage
    module: raspberrypi
    pin: 13
    on_payload: "ON"
    off_payload: "OFF"
    initial: high  # This optional value controls the initial state of the pin before receipt of any messages from MQTT. Valid options are 'low' and 'high'.
    inverted: yes  # This pin may control an open-collector output which is "on" when the output is "low".
    retain: yes # This option value controls if the message is retained. Default is no.

sensor_modules:
  - name: dht22_outside
    module: dht22
    type: DHT22 # can be  DHT11, DHT22 or AM2302
    pin: 15

  - name: dht22_inside
    module: dht22
    type: DHT22 # can be  DHT11, DHT22 or AM2302
    pin: 4

sensor_inputs:
  - name: garageoutsidetemperature
    module: dht22_outside
    interval: 60 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 4 # number of digits to be round
    type: temperature # Can be temperature or humidity
      
  - name: garageoutsidehumidity 
    module: dht22_outside
    interval: 60 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 4 # number of digits to be round
    type: humidity # Can be temperature or humidity   

  - name: garageinsidetemperature
    module: dht22_inside
    interval: 60 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 4 # number of digits to be round
    type: temperature # Can be temperature or humidity
      
  - name: garageinsidehumidity 
    module: dht22_inside
    interval: 60 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 4 # number of digits to be round
    type: humidity # Can be temperature or humidity   

digital_inputs:
  - name: garagedoorsensor
    module: raspberrypi
    pin: 22
    on_payload: "ON"
    off_payload: "OFF"
    pullup: no
    pulldown: yes

  - name: gardendoorsensor
    module: raspberrypi
    pin: 25
    on_payload: "ON"
    off_payload: "OFF"
    pullup: no
    pulldown: yes

  - name: gardenlightsensor
    module: raspberrypi
    pin: 21
    on_payload: "ON"
    off_payload: "OFF"
    pullup: no
    pulldown: yes

  - name: gardenmotionsensor
    module: raspberrypi
    pin: 17
    on_payload: "OFF"
    off_payload: "ON"
    pullup: no
    pulldown: yes 
```

closed 2021-07-13 by mihadoo