flyte/mqtt-io #235: [Request] PIR and Flame Module Sensor

1722007, 2021-09-14:
**PIR sensor**
The PIR sensor itself has two slots in it, each slot is made of a special material that is sensitive to IR. The lens used here is not really doing much and so we see that the two slots can 'see' out past some distance (basically the sensitivity of the sensor). When the sensor is idle, both slots detect the same amount of IR, the ambient amount radiated from the room or walls or outdoors. When a warm body like a human or animal passes by, it first intercepts one half of the PIR sensor, which causes a positive differential change between the two halves. When the warm body leaves the sensing area, the reverse happens, whereby the sensor generates a negative differential change. These change pulses are what is detected.
(https://learn.adafruit.com/pir-passive-infrared-proximity-motion-sensor/how-pirs-work)

**Flame module sensor**
A flame sensor module that consists of a flame sensor (IR receiver), resistor, capacitor, potentiometer, and comparator LM393 in an integrated circuit. It can detect infrared light with a wavelength ranging from 700nm to 1000nm. The far-infrared flame probe converts the light detected in the form of infrared light into current changes. Sensitivity is adjusted through the onboard variable resistor with a detection angle of 60 degrees. Working voltage is between 3.3v and 5.2v DC, with a digital output to indicate the presence of a signal. Sensing is conditioned by an LM393 comparator.
(https://www.pcboard.ca/flame-sensor-module)

**Benefits**
Those hardwares would be a great addition for the security and fire detection system respectively for a smart home.

**Context**
Currently those cannot be used? Are there any alternatives that can replace those 2 if those 2 won't be implemented here?

koleo9am, 2021-10-04:
Both modules, you ask for support, have digital output. There is no need to add special support for any of it. 
Just configure it as a digital input of suitable GPIO module.

BenjiU, 2022-11-04:
No new module required, as koleo9am showed.

closed 2022-11-04 by BenjiU