flyte/mqtt-io #241: [HA Discovery] Add state_class: measurement to config. It will be fix graphs and storing

a-x-, 2021-11-07:
**Describe the bug**

a) mqtt-io graph:
img

b) correct graph:
img

**Config**
```yaml
sensor_modules:
  - name: wc_climate_bme280
    module: bme280
    i2c_bus_num: 1
    chip_addr: 0x76

sensor_inputs:
  - name: wc_humidity
    module: wc_climate_bme280
    type: humidity
    retain: yes
    extra_announce_payload:
      device_class: "humidity"
```

**Hardware**
 - Platform: Raspberry Pi
 - Connected hardware: does not matter. In my case: bme280

**System:** does not matter

**Additional context**
https://developers.home-assistant.io/docs/core/entity/sensor/#available-state-classes

MrC0nst, 2021-11-12:
Is it possible to add "unit_of_measurement" for sensors with HA autodiscovery? 
Temperature sensor example(from aqara temperature and humidity sensor):
"unit_of_measurement": "°C", 
"state_class": "measurement",

Humidity sensor example:
"unit_of_measurement": "%",
"state_class": "measurement",

This trick will add easy-reading values to HA interface
[image]

stokket, 2021-12-02:
I managed to enable this by changing following in config.yml

sensor_modules:
  - name: Temp1
    module: ds18b
    type: DS18B20
    address: 000004cd45ba

sensor_inputs:
  - name: Temp1
    module: ds18b20
    interval: 10
    digits: 1
    ha_discovery:
      name: Temp1
      device_class: temperature
      unit_of_measurement: °C

MrC0nst, 2021-12-02:
> I managed to enable this by changing following in config.yml
>
> sensor_modules:
>
> * name: Temp1
>   module: ds18b
>   type: DS18B20
>   address: 000004cd45ba
>
> sensor_inputs:
>
> * name: Temp1
>   module: [...]
 Really good solution, but i suggest to use quotes unit_of_measurement: "°C" for temperature or unit_of_measurement: "%" for humidity. Otherway you will have error _raise ScannerError("while scanning for the next token", None, yaml.scanner.ScannerError: while scanning for the next token found character '%' that cannot start any token_

OzGav, 2022-04-28:
hey OP can you close this because it is already possible?

leochras, 2022-07-09:
I'm getting:
 `mqtt_io.exceptions.ConfigValidationFailed: Config did not validate:
unit_of_measurement:
- unknown field`
Is this feature enabled or not?

lucianojss, 2022-11-20:
As stated before you can simply do:

```yml
sensor_modules:
  - name: ds18b_module
    module: ds18b
    type: DS18B20
    address: 0000057b6ebc

sensor_inputs:
  - name: temp_sensor
    module: ds18b_module
    interval: 30
    ha_discovery:
      name: Temperature sensor
      device_class: temperature
      unit_of_measurement: °C
```