flyte/mqtt-io #243: Connection reset by peer

Julusian, 2021-12-06:
- [x ] I have read the [README.md](https://github.com/flyte/pi-mqtt-gpio/blob/develop/README.md) file and visited the [Documentation](https://mqtt-io.app/) to see if the information there helps.

**Describe the bug**
Not reconnecting to MQTT

**Expected behavior**
The software to reconnect to the MQTT server, or exit with a fatal error

**Error messages and traceback**
```
2021-12-06 00:24:39 mqtt_io.server [INFO] Digital output 'relay11' set to False (off)
2021-12-06 00:24:39 mqtt_io.server [INFO] Digital output 'relay12' set to False (off)
failed to receive on socket: [Errno 104] Connection reset by peer
```

**Config**
```yaml
mqtt:
  host: 
  port: 1883
  user: ""
  password: ""
  topic_prefix: relay/rack

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_outputs:
  - name: relay1
    module: raspberrypi
    pin: 2
    on_payload: "ON"
    off_payload: "OFF"

```
More outputs are defined, in an identical fashion

**Hardware**
 - Platform: Raspberry Pi 1 Model B 512MB 
 - Connected hardware:  N/A

**System:**
 - OS: Raspbian
 - Python version: 3.7.3
 - User you're running as: pi
 - Using a virtualenv?: no

**Additional context**
I believe this is also happening to some pi zeros too, but I do not have the logs to verify

CoolSaet, 2022-01-25:
I have the same issue. However no error message. 
I am also running only digital outputs. On an RPI 2B with bullseye installed. Everything works fine with mosquito server running as addon in home Assistant. After HA reboot or restart of the mosquito addon, nothing is received on the mqtt-io client. Only restarting the process seems to fix it. Basically reinitiating the connection.

Bettman66, 2022-01-25:
The problem has existed since the switch to asynchronous. You have to control an input first and it reconnects. I went back to the old version, it works great.

CoolSaet, 2022-01-25:
> The problem has existed since the switch to asynchronous. You have to control an input first and it reconnects. I went back to the old version, it works great.

Thanks. Could you tell me which version is the old one? Everything below 2.0.0?

Bettman66, 2022-01-25:
I forked and removed everything but PIFace and GPIO. If that is enough for you, you can use this. I added installation instructions in German. You can use the translator.

Julusian, 2022-01-25:
My solution to this issue has been to switch to use an esp32 running esphome (connecting to home assistant)