flyte/mqtt-io #255: Installation problem (novice)

veqlargh, 2022-02-08:
Hi. I am new to this kind of stuff. I have installed HA to my RPI3 and i want to hardwire my sensors and relay to RPI3. I came across the mqtt-io. 
I have an installation problem to start with. I have done this installation
`pip3 install mqtt-io`
and it worked. 
when i tried to execute next 
`python3 -m mqtt_io config.yml`
I got an error
`FileNotFoundError: [Errno 2] No such file or directory: 'config.yml'
`
Do I have to create a config.yml file myself - if yes, where do i create it? Is there something else I should do before? 
Or is there maybe some tutorial on how to do this stuff.
Sorry for so many questions but I would really like to learn :)

BenjiU, 2022-02-08:
Hi, there are some config examples in the repo ([config.example.yml](https://github.com/flyte/mqtt-io/blob/develop/config.example.yml)). Feel free to use the needed information for your hardware and create your own config.yml
Locate it, where it fits for you. e.g. pi home directory "/home/pi/config.yml". Depending on the user you're running mqtt_io, you should give the full path of the config file. `python3 -m mqtt_io /home/pi/config.yml`

veqlargh, 2022-02-08:
Thank you for expelling.

Now I ran into another issue. When I execute the config file I get one of the errors. 

If i add a 
```
gpio_modules:
  - name: MyPI
    module: raspberrypi
```

to the config file I get this error:

```
Config did not validate due to errors in the following sections:
gpio_modules:
  MyPI:
  - GPIO module 'MyPI' does not have any digital_input or digital_output configs
```

But if I remove upper config lines I get error as below. Am I doing something wrong again? The config file is below the error.

```
Collecting Adafruit_DHT
  Using cached Adafruit_DHT-1.4.0.tar.gz (15 kB)
Using legacy 'setup.py install' for Adafruit-DHT, since package 'wheel' is not installed.
Installing collected packages: Adafruit-DHT
    Running setup.py install for Adafruit-DHT ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-j6qvzo3w/adafruit-dht_1dd1101df8614e2cb15ca01d9f26f982/setup.py'"'"'; __file__='"'"'/tmp/pip-install-j6qvzo3w/adafruit-dht_1dd1101df8614e2cb15ca01d9f26f982/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ex1swoia/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/Adafruit-DHT
         cwd: /tmp/pip-install-j6qvzo3w/adafruit-dht_1dd1101df8614e2cb15ca01d9f26f982/
    Complete output (20 lines):
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-aarch64-3.9
    creating build/lib.linux-aarch64-3.9/Adafruit_DHT
    copying Adafruit_DHT/platform_detect.py -> build/lib.linux-aarch64-3.9/Adafruit_DHT
    copying Adafruit_DHT/Raspberry_Pi_2.py -> build/lib.linux-aarch64-3.9/Adafruit_DHT
    copying Adafruit_DHT/Raspberry_Pi.py -> build/lib.linux-aarch64-3.9/Adafruit_DHT
    copying Adafruit_DHT/common.py -> build/lib.linux-aarch64-3.9/Adafruit_DHT
    copying Adafruit_DHT/Beaglebone_Black.py -> build/lib.linux-aarch64-3.9/Adafruit_DHT
    copying Adafruit_DHT/Test.py -> build/lib.linux-aarch64-3.9/Adafruit_DHT
    copying Adafruit_DHT/__init__.py -> build/lib.linux-aarch64-3.9/Adafruit_DHT
    running build_ext
    building 'Adafruit_DHT.Raspberry_Pi_2_Driver' extension
    creating build/temp.linux-aarch64-3.9
    creating build/temp.linux-aarch64-3.9/source
    creating build/temp.linux-aarch64-3.9/source/Raspberry_Pi_2
    gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -g -fno-semantic-interposition -g -fno-semantic-interposition -g -fno-semantic-interposition -DTHREAD_STACK_SIZE=0x100000 -fPIC -I/usr/include/python3.9 -c source/Raspberry_Pi_2/pi_2_dht_read.c -o build/temp.linux-aarch64-3.9/source/Raspberry_Pi_2/pi_2_dht_read.o -std=gnu99
    error: command 'gcc' failed: No such file or directory
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-j6qvzo3w/adafruit-dht_1dd1101df8614e2cb15ca01d9f26f982/setup.py'"'"'; __file__='"'"'/tmp/pip-install-j6qvzo3w/adafruit-dht_1dd1101df8614e2cb15ca01d9f26f982/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ex1swoia/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/Adafruit-DHT Check the logs for full command output.
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/mqtt_io/modules/__init__.py", line 51, in install_missing_module_requirements
    install_missing_requirements(pkgs_required)
  File "/usr/lib/python3.9/site-packages/mqtt_io/modules/__init__.py", line 22, in install_missing_requirements
    check_call([sys.executable, "-m", "pip", "install"] + pkgs_required)
  File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'install', 'Adafruit_DHT']' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
[... 22 lines cut ...]
  topic_prefix: topic
  
gpio_modules:
  - name: MyPI
    module: raspberrypi

sensor_modules:
  - name: SensorTempHumid
    module: dht22
    type: DHT22
    pin: 4

sensor_inputs:
  - name: temperature
    module: SensorTempHumid
    interval: 10
    digits: 4
    type: temperature

  - name: humidity
    module: SensorTempHumid
    interval: 10
    digits: 4
    type: humidity
```

BenjiU, 2022-02-10:
Hey,
- the first problem was solved by you (mqtt_io does not like unused modules.
- your second problem:
mqtt_io tries to install the dht22 lib, but has problems:
- Using legacy 'setup.py install' for Adafruit-DHT, since package 'wheel' is not installed.
When wheel is not installed, it tries to build it on its own, but gcc is missing too, so it fails.
Maybe you can try `pip install wheel` manually and try to run mqtt_io again.
Good luck

closed 2022-04-04 by BenjiU