flyte/mqtt-io #259: configure problem - novice!

mattb1908, 2022-03-30:
Hi
running on raspian, 

on execute I get the following:

python3 -m mqtt_io config.MCP3008.yml
2022-03-31 00:53:17 mqtt_io.__main__ [ERROR] MqttIo crashed!
Traceback (most recent call last):
  File "/home/openhabian/mqtt-io/mqtt_io/__main__.py", line 104, in main
    mqtt_gpio.run()
  File "/home/openhabian/mqtt-io/mqtt_io/server.py", line 1229, in run
    self._init_sensor_modules()
  File "/home/openhabian/mqtt-io/mqtt_io/server.py", line 255, in _init_sensor_modules
    self.sensor_modules[sens_config["name"]] = _init_module(
  File "/home/openhabian/mqtt-io/mqtt_io/server.py", line 128, in _init_module
    return module_class(module_config)
  File "/home/openhabian/mqtt-io/mqtt_io/modules/sensor/__init__.py", line 22, in __init__
    self.setup_module()
  File "/home/openhabian/mqtt-io/mqtt_io/modules/sensor/mcp3008.py", line 46, in setup_module
    spi=SPI.SpiDev(self.config["spi_port"], self.config["spi_device"])
  File "/home/openhabian/.local/lib/python3.9/site-packages/Adafruit_GPIO/SPI.py", line 42, in __init__
    self._device.open(port, device)
FileNotFoundError: [Errno 2] No such file or directory
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/openhabian/mqtt-io/mqtt_io/__main__.py", line 112, in <module>
    main()
  File "/home/openhabian/mqtt-io/mqtt_io/__main__.py", line 104, in main
    mqtt_gpio.run()
  File "/home/openhabian/mqtt-io/mqtt_io/server.py", line 1229, in run
    self._init_sensor_modules()
  File "/home/openhabian/mqtt-io/mqtt_io/server.py", line 255, in _init_sensor_modules
    self.sensor_modules[sens_config["name"]] = _init_module(
  File "/home/openhabian/mqtt-io/mqtt_io/server.py", line 128, in _init_module
    return module_class(module_config)
  File "/home/openhabian/mqtt-io/mqtt_io/modules/sensor/__init__.py", line 22, in __init__
    self.setup_module()
  File "/home/openhabian/mqtt-io/mqtt_io/modules/sensor/mcp3008.py", line 46, in setup_module
    spi=SPI.SpiDev(self.config["spi_port"], self.config["spi_device"])
  File "/home/openhabian/.local/lib/python3.9/site-packages/Adafruit_GPIO/SPI.py", line 42, in __init__
    self._device.open(port, device)
FileNotFoundError: [Errno 2] No such file or directory

configure file is :

mqtt:
  host: 192.168.##.##
  port: 1883
  user: ""
  password: ""
  client_id: control_panel
  clean_session: true
  topic_prefix: battery
  status_payload_running: "online"
  status_payload_stopped: "offline"
  status_payload_dead: "offline"

sensor_modules:
  - name: mcp_adc
    module: mcp3008
    
    
sensor_inputs:
  - name: test_adc
    module: mcp_adc
    channel: CH0

any insight appreciated...

BenjiU, 2022-03-31:
Hi,

somethings not working with your SPI:
```
File "/home/openhabian/.local/lib/python3.9/site-packages/Adafruit_GPIO/SPI.py", line 42, in init
self._device.open(port, device)
FileNotFoundError: [Errno 2] **No such file or directory**
```

As far as I remember, you have to switch on SPI on the raspberry. e.g. via `sudo raspi-config`

mattb1908, 2022-03-31:
Thanks, will try that this eve........

mattb1908, 2022-04-02:
Thanks, SPI enabled and its now working. 
I have a mcp3008 and using this config i am getting the reading from channel 0 every 15 secs.
I am struggling to to set the channel number, can be 0 - 7 

mqtt:
  host: 192.168.0.10
  port: 1883
  user: ""
  password: ""
  topic_prefix: battery

gpio_modules:
  - name: rpi
    module: raspberrypi 

sensor_modules:
  - name: mcp3008
    module: mcp3008

sensor_inputs:
  - name: analog7
    module: mcp3008
    interval: 15

however f I add the channel number:

sensor_inputs:
  - name: analog7
    module: mcp3008
    channel: 7
    interval: 15

i get the following:

openhabian@openhabian:~ $ python -m pi_mqtt_gpio.server mqtt-io/config.MCP3008.yml                    2022-04-02 21:43:03,338 mqtt_gpio (INFO): Startup
2022-04-02 21:43:03,604 mqtt_gpio (ERROR): Config for 'mcp3008' sensor named 'analog7' did not validate:
channel:
- must be of string type

looking at the schema:

    SENSOR_SCHEMA = {
        "channel": dict(
            type="integer",
            required=True,
            min=0,
            max=7,
        )
    }

mattb1908, 2022-04-19:
solved this by using the following
I can now specify channels.

sensor_inputs:
  - name: analog0
    module: mcp3008
    interval: 10
    channel: CH0

closed 2022-04-19 by mattb1908