flyte/mqtt-io #260: Can't get it to work. Error from python?

luvxinh, 2022-04-02:
- [Y] I have read the [README.md](https://github.com/flyte/pi-mqtt-gpio/blob/develop/README.md) file and visited the [Documentation](https://mqtt-io.app/) to see if the information there helps.
- Also have gone through the closed issues, but I don't have same errors.

**Describe the bug**
After installing python, I create the yaml file.

`pi@pi21:~ $ sudo nano test-config.yml`

with the following content

`mqtt:
  host: core-mosquitto
  port: 1883
  user: "mqtt"
  password: "password"
  topic_prefix: /home/temp

# GPIO
gpio_modules:
  # Use the Raspberry Pi built-in GPIO
  - name: rpi
    module: raspberrypi

digital_inputs:
  # Pin 0 is an input connected to a doorbell button
  - name: Chun
    module: rpi
    pin: 4

sensor_modules:
  - name: dht22
    module: dht22
    type: DHT22 # can be  DHT11, DHT22 or AM2302
    pin: 4

sensor_inputs:
  - name: temperatureChun
    module: dht22
    interval: 5 #interval in seconds, that a value is read from the senso>
    digits: 4 # number of digits to be round
    type: temperature # Can be temperature or humidity

  - name: humidityChun
    module: dht22
    interval: 5 #interval in seconds, that a value is read from the senso>
    digits: 4 # number of digits to be round
    type: humidity # Can be temperature or humidity
`
It's basically just to get the data off the DHT22 sensor.

However, when I run

`
pi@pi21:~ $ python3 -m mqtt_io test-config.yml
`

the following errors appear:

`
2022-04-02 11:08:02 mqtt_io.server [INFO] Connecting to MQTT...
2022-04-02 11:08:02 mqtt_io.server [INFO] Digital input 'Chun' value changed to True
2022-04-02 11:08:02 mqtt_io.server [INFO] Read sensor 'humidityChun' value of 55.4
2022-04-02 11:08:02 mqtt_io.server [INFO] Read sensor 'temperatureChun' value of 27.0
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/asyncio_mqtt/client.py", line 79, in connect
    await loop.run_in_executor(None, self._client.connect, self._hostname, self._port, 60)
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/paho/mqtt/client.py", line 914, in connect
    return self.reconnect()
  File "/home/pi/.local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1044, in reconnect
    sock = self._create_socket_connection()
  File "/home/pi/.local/lib/python3.9/site-packages/paho/mqtt/client.py", line 3685, in _create_socket_connection
    return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)
  File "/usr/lib/python3.9/socket.py", line 822, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
  File "/usr/lib/python3.9/socket.py", line 953, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/mqtt_io/mqtt/asyncio_mqtt.py", line 32, in inner
    await func(*args, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/mqtt_io/mqtt/asyncio_mqtt.py", line 79, in connect
    await self._client.connect(timeout=timeout)
  File "/home/pi/.local/lib/python3.9/site-packages/asyncio_mqtt/client.py", line 88, in connect
    raise MqttError(str(error))
asyncio_mqtt.error.MqttError: [Errno -2] Name or service not known

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/mqtt_io/server.py", line 1158, in _main_loop
    await self._connect_mqtt()
  File "/home/pi/.local/lib/python3.9/site-packages/mqtt_io/server.py", line 585, in _connect_mqtt
    await self.mqtt.connect()
  File "/home/pi/.local/lib/python3.9/site-packages/mqtt_io/mqtt/asyncio_mqtt.py", line 34, in inner
    raise MQTTException from exc
mqtt_io.mqtt.MQTTException

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/pi/.local/lib/python3.9/site-packages/mqtt_io/__main__.py", line 85, in <module>
    main()
  File "/home/pi/.local/lib/python3.9/site-packages/mqtt_io/__main__.py", line 80, in main
    mqtt_gpio.run()
  File "/home/pi/.local/lib/python3.9/site-packages/mqtt_io/server.py", line 1237, in run
    self.loop.run_until_complete(self._main_task)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/pi/.local/lib/python3.9/site-packages/mqtt_io/server.py", line 1183, in _main_loop
    if reconnects_remaining is not None:
UnboundLocalError: local variable 'reconnects_remaining' referenced before assignment
sys:1: RuntimeWarning: coroutine 'MqttIo._mqtt_publish' was never awaited
`
I just want it to run and publish something. I'm listening on homeassistant but there's just nothing

**Hardware**
 - Platform: Raspberry Pi 4
 - Connected hardware: DHT22.

**System:**
 - OS: Raspbian Bulleyes
 - Python version: 3.9.2
 - User you're running as: pi  
 - Using a virtualenv?: no

**Additional context**
N/A

dantrainor, 2022-05-01:
Are you able to use mosquitto_sub and mosquitto_pub test clients using the information you provided for the MQTT server in your configuration?

BenjiU, 2022-11-04:
closed due to no response

closed 2022-11-04 by BenjiU