flyte/mqtt-io #262: RPI0 2W : GPIO 5 always at HIGH level when mqtt-io start

xNiux, 2022-04-04:
- [ ] I have read the [README.md](https://github.com/flyte/pi-mqtt-gpio/blob/develop/README.md) file and visited the [Documentation](https://mqtt-io.app/) to see if the information there helps.

**Describe the bug**
When MQTT-IO service start, it set to high the GPIO 5.

**Expected behavior**
In config, I set initial state to low. But when mqtt-io start, il set to high the GPIO5.
If I publish a message OFF on GPIO5, it fall to low. 
Currently, I only have GPIO5 and 6 that are connected. The GPIO6 is OK.

**Error messages and traceback**
```
-
```

**Config**
```yaml
mqtt:
  host: localhost
  port: ***
  topic_prefix: home/poulailler
  ha_discovery:
    enabled: yes

# GPIO
gpio_modules:
  # Use the Raspberry Pi built-in GPIO
  - name: rpi
    module: raspberrypi

digital_inputs:
  # Pin 20 is an input connected to a switch on Poulailler door
  - name: poulailler_door_switch_closed
    module: rpi
    pin: 20

  # Pin 21 is an input connected to a switch on Poulailler door
  - name: poulailler_door_switch_opened
    module: rpi
    pin: 21

digital_outputs:
  # Pin 5 is an output connected to IR Right of Poulailler Camera
  - name: poulailler_camera_ir_right
    module: rpi
    pin: 5
    initial: low

  # Pin 6 is an output connected to IR Left  of Poulailler Camera.
  - name: poulailler_camera_ir_left
    module: rpi
    pin: 6
    initial: low

  # Pin 26 is an output connected to a motor on Poulailler door
  - name: poulailler_door_motor
    module: rpi
    pin: 26
    initial: low

  # Pin 12 is an output connected to a cooling fan (PWM Output)
  - name: poulailler_cooling_fan
    module: rpi
    pin: 12
    initial: low

```

**Hardware**
 - Platform: Raspberry Zero 2 W
 - Connected hardware:  RPI Camera

**System:**
 - OS: Raspberry Debian 11 Bullseye
 - Python version: 3.9.2
 - User you're running as: pi
 - Using a virtualenv?: no

dantrainor, 2022-05-01:
Pins 0-8 are always set HIGH by default in the SoC (they are input only and the idea is, the input needs a voltage). You can set a config.txt to set them low and this will take effect after the bootloader runs but the system will still start with them HIGH by default, every time. I'd suggest using different pins.

xNiux, 2022-05-01:
Rugby! It's clear. Thanks a lot

closed 2022-05-01 by xNiux