flyte/mqtt-io #265: Input/ Output status after HomeAssistant Reboot

eknower, 2022-05-05:
Hi, I'm using MQTT-IO on RPI to expose inputs & outputs to Home Assistant.
Problem is, that after a reboot of Home Assistant OR of the Pi, all of my inputs and outputs statuses are unknown.
This prevents my automations from working correctly until every certain input / output is being triggered for the first time, which then corrects it's state.

Any solution? (I'm also open for any implementation of automation on reboot, but then I'll have to figure out how to run it while HA reboot AND while Pi reboots, just in case)

Appreciate any help!

eknower, 2022-05-08:
It seems that the issue occurs only while home assistant is restarting while the Pi keeps running.
Therefore my solution was to implement an automation which restarts the Pi after Home Assistant starts.
Seems to work perfectly.
Implementation was done using a bash script + ssh access using a key.

closed 2022-05-08 by eknower

RaVioLb, 2022-05-09:
You need to set the option retain to true. With this option i not have any trouble with reboot HA