flyte/mqtt-io #266: hcsr04 config validation issue

techadrian, 2022-05-07:
Been trying to use a ultrasonic sesnsor hcsr04 .
I have build the docker container on my rpi3 with the latest code fri develop branch today (07-05-2022).
I  have conencted the sensor and use your exact configuation example from here : https://hub.docker.com/r/flyte/mqtt-gpio
eq : 

sensor_modules:

geccko, 2022-07-07:
Hello,

I have the same problem with hcsr04. Do you have a solution??

  self.io.setup(pin, direction, pull_up_down=pullup, initial=initial_int)
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/pi/.local/lib/python3.9/site-packages/mqtt_io/__main__.py", line 85, in <module>
    main()
  File "/home/pi/.local/lib/python3.9/site-packages/mqtt_io/__main__.py", line 80, in main
    mqtt_gpio.run()
  File "/home/pi/.local/lib/python3.9/site-packages/mqtt_io/server.py", line 1229, in run
    self._init_sensor_modules()
  File "/home/pi/.local/lib/python3.9/site-packages/mqtt_io/server.py", line 255, in _init_sensor_modules
    self.sensor_modules[sens_config["name"]] = _init_module(
  File "/home/pi/.local/lib/python3.9/site-packages/mqtt_io/server.py", line 122, in _init_module
    module_config = validate_and_normalise_config(module_config, module_schema)
  File "/home/pi/.local/lib/python3.9/site-packages/mqtt_io/config/__init__.py", line 105, in validate_and_normalise_config
    raise ConfigValidationFailed(
mqtt_io.exceptions.ConfigValidationFailed: Config did not validate:
burst:
- unknown field
pin_echo:
- unknown field
pin_trigger:
- unknown field

thanks!

geccko, 2022-07-10:
@techadrian did you solved your problem ?

techadrian, 2022-07-13:
Hello @geccko  ,
No. Nobody responded for several months and I do not have the python skills to find a way out of this. 
I decided to move to another project since flyte/mqtt-io looks dead. 
Also this project has other big issues for me, like the fact that after two-three days or running , the relay states and relay commands take over 1 minute to execute/read (I am using a docker container to run this app ) so I need to restart the app every night. Also....hw resource consumption is terrible on my rpi3b ....

geccko, 2022-07-13:
Hello @techadrian,

I'm testing other project which I modified for my needs and I'm testing the ultrasonic sensor with good results. Which project are you using? please let me know. 

Thanks

techadrian, 2022-07-14:
Hello @geccko ,

I order to simplify things, I started modfying https://github.com/alaub81/rpi_sensor_scripts , created a bash script that will execute those python scripts to get the data  and pass it to home assisntant, also containarised the whole thing using docker.  I kinda started with https://github.com/alaub81/rpi_sensor_scripts but now I see  it is slowly evolving into a new solution :)

andrewradin, 2024-12-28:
After inspecting the code, and looking at the provided example, the problem is the example configuration referenced is not correct. Pin, trigger and burst are all sensor configuration, not module configuration. The configuration should look something more like the example below. I can confirm a configuration of this type does work in my environment. 

sensor_modules:
  - name: hcsr04_sensor
    module: hcsr04

sensor_inputs:
  - name: hcsr04_distance
    module: hcsr04_sensor
    pin_echo: 27
    pin_trigger: 17
    burst: 10
    interval: 10
    digits: 1