flyte/mqtt-io #273: No outputs publishing to MQTT broker

bsmith76s, 2022-06-11:
**Action description**
I am not getting the digital outputs publishing to broker. I used MQTT Explorer and see nothing on outputs but inputs are publishing

**Previous attempts**
I have minimized the digital outputs sections to barely anything with no success. 

**Config**
```mqtt:
  host: xx.xx.xx.xx
  port: 1883
  user: "mqtt"
  password: "xxxxx"
  topic_prefix: pimqtt/house_garage
  status_topic: status
  status_payload_running: online
  status_payload_stopped: offline
  status_payload_dead: dead

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_outputs:
  - name: door1
    module: raspberrypi
    pin: 5
    on_payload: "ON"
    off_payload: "OFF"
    inverted: true
    initial: high
#  - name: door2
#    module: raspberrypi
#    pin: 6 #GPIO 
#    on_payload: "ON"
#    off_payload: "OFF"
#    inverted: true
#    initial: high
digital_inputs:
  - name: door1_closed
    module: raspberrypi
    pin: 22 # GPIO 
    on_payload: "ON"
    off_payload: "OFF"
    pullup: yes
    pulldown: no
```

**Hardware**
 - Platform: Raspberry Pi 0W
 - Connected hardware:  2 channel relay, reed sensors

**System:**
 - OS: [Raspberry Pi OS Bullseye]
 - Python version: [3.9`]
 - User you're running as: root
 - Using a virtualenv?: yes

bsmith76s, 2022-06-11:
closed figure it out

bsmith76s, 2022-06-11:
figure it out closed

closed 2022-06-11 by bsmith76s

day1118, 2023-05-01:
I am facing the same issue @bsmith76s - what was the solution?

bsmith76s, 2023-05-01:
> I am facing the same issue @bsmith76s - what was the solution?

digital outputs are only published to broker when you publish a message to them. I used home assistant to publish a message to the sensor and it showed in MQTT explorer.