flyte/mqtt-io #276: mqtt connected and working. mcp23017 config failure. unable to control gpio module

BoW2EviL, 2022-07-05:
- [X] I have read the [README.md](https://github.com/flyte/pi-mqtt-gpio/blob/develop/README.md) file and visited the [Documentation](https://mqtt-io.app/) to see if the information there helps.

**Action description**
A clear and concise description of what you're trying to achieve.

- i am attempting to use mcp23017 with mqtt-io

**Previous attempts**
Describe what you've tried so far.
- i can get mqtt-io to start and connect to the mqtt broker.
- using mqtt explorer i attempted to manually trigger the output with a raw ON | OFF post to "home/livingroom/climate/output/socket8/set" with success, mqtt-io responds by setting home/livingroom/climate/output/socket8 to ON | OFF respectively. 

- i seem to get no debug logs concerning mcp23017 so i am not sure how to determine whether or not mcp23017 module is doing?  
- i2c_bus_num fails validation when enabled with gpio module mcp23017 like the examples show

**Error messages and traceback**
```
python3 -m mqtt_io config.yaml
2022-07-05 20:15:37 mqtt_io.modules [DEBUG   ] Module <module 'mqtt_io.modules.gpio.mcp23017' from '/usr/local/lib/python3.9/dist-packages/mqtt_io/modules/gpio/mcp23017.py'> has all of its requirements installed already.
2022-07-05 20:15:37 mqtt_io.server [DEBUG   ] Going Asynchronous
2022-07-05 20:15:38 mqtt_io.server [INFO    ] Connecting to MQTT...
2022-07-05 20:15:38 mqtt_io.server [INFO    ] Connected to MQTT
2022-07-05 20:15:38 mqtt_io.server [DEBUG   ] Running MQTT task with priority 1: <coroutine object MqttIo._mqtt_subscribe at 0x75bbab28>
2022-07-05 20:15:38 mqtt_io.server [INFO    ] Subscribed to topic: 'home/livingroom/climate/output/socket14/set'
2022-07-05 20:15:38 mqtt_io.server [INFO    ] Subscribed to topic: 'home/livingroom/climate/output/socket14/set_on_ms'
2022-07-05 20:15:38 mqtt_io.server [INFO    ] Subscribed to topic: 'home/livingroom/climate/output/socket14/set_off_ms'
2022-07-05 20:15:38 mqtt_io.server [DEBUG   ] Running MQTT task with priority 3: <coroutine object MqttIo._mqtt_publish at 0x75cd1ce8>
2022-07-05 20:15:38 mqtt_io.server [DEBUG   ] Publishing MQTT message on topic 'home/livingroom/climate/status': 'running'
2022-07-05 20:15:58 mqtt_io.server [DEBUG   ] Received message on topic 'home/livingroom/climate/output/socket14/set': 'ON'
2022-07-05 20:15:58 mqtt_io.server [INFO    ] Digital output 'socket14' set to True (on)
2022-07-05 20:15:58 mqtt_io.events [DEBUG   ] Found 1 listener(s) for event type DigitalOutputChangedEvent
2022-07-05 20:15:58 mqtt_io.server [DEBUG   ] Running MQTT task with priority 3: <coroutine object MqttIo._mqtt_publish at 0x75aee4a8>
2022-07-05 20:15:58 mqtt_io.server [DEBUG   ] Publishing MQTT message on topic 'home/livingroom/climate/output/socket14': 'ON'
2022-07-05 20:16:04 mqtt_io.server [DEBUG   ] Received message on topic 'home/livingroom/climate/output/socket14/set': 'OFF'
2022-07-05 20:16:04 mqtt_io.server [INFO    ] Digital output 'socket14' set to False (off)
2022-07-05 20:16:04 mqtt_io.events [DEBUG   ] Found 1 listener(s) for event type DigitalOutputChangedEvent
2022-07-05 20:16:04 mqtt_io.server [DEBUG   ] Running MQTT task with priority 3: <coroutine object MqttIo._mqtt_publish at 0x75aee528>
2022-07-05 20:16:04 mqtt_io.server [DEBUG   ] Publishing MQTT message on topic 'home/livingroom/climate/output/socket14': 'OFF'

shutdown with ^C
2022-07-05 20:16:33 mqtt_io.server [DEBUG   ] Clearing events and cancelling 'critical_tasks'
2022-07-05 20:16:33 mqtt_io.server [DEBUG   ] 'critical_tasks' cancelled and gathered
2022-07-05 20:16:33 mqtt_io.server [INFO    ] Waiting for our tasks to complete...
2022-07-05 20:16:33 mqtt_io.server [DEBUG   ] Publishing MQTT message on topic 'home/livingroom/climate/status': 'stopped'
2022-07-05 20:16:33 mqtt_io.server [INFO    ] Disconnecting from MQTT...
2022-07-05 20:16:33 mqtt_io.server [INFO    ] MQTT disconnected
2022-07-05 20:16:33 mqtt_io.server [DEBUG   ] Loop closed
2022-07-05 20:16:33 mqtt_io.server [DEBUG   ] Running cleanup on module <mqtt_io.modules.gpio.mcp23017.GPIO object at 0x75cf3760>
2022-07-05 20:16:33 mqtt_io.server [DEBUG   ] run() complete

```
**traceback failure with the i2c_bus_num enabled**
```
2022-07-05 19:05:24 mqtt_io.modules [DEBUG   ] Module <module 'mqtt_io.modules.gpio.mcp23017' from '/usr/local/lib/python3.9/dist-packages/mqtt_io/modules/gpio/mcp23017.py'> has all of its requirements installed already.
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/__main__.py", line 85, in <module>
    main()
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/__main__.py", line 80, in main
    mqtt_gpio.run()
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/server.py", line 1228, in run
    self._init_digital_outputs()
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/server.py", line 450, in _init_digital_outputs
    out_conf = validate_and_normalise_digital_output_config(out_conf, gpio_module)
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/config/__init__.py", line 226, in validate_and_normalise_digital_output_config
    return validate_and_normalise_config(
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/config/__init__.py", line 105, in validate_and_normalise_config
[... 110 lines cut ...]
pyparsing                        3.0.9
pyserial                         3.5
python-apt                       2.2.1
python-periphery                 2.3.0
pyusb                            1.2.1
PyYAML                           5.4.1
requests                         2.25.1
rpi-ws281x                       4.3.4
RPi.GPIO                         0.7.0
setuptools                       52.0.0
six                              1.16.0
smbus2                           0.4.2
spidev                           3.5
ssh-import-id                    5.10
sysv-ipc                         1.1.0
typing                           3.7.4.3
typing-extensions                3.10.0.2
urllib3                          1.26.5
wheel                            0.34.2
yamlloader                       0.5.5
zipp                             3.8.0

```

Thanks for any advise you can give.

bsmith76s, 2022-07-05:
try this chip address

 chip_addr: 0x27

BoW2EviL, 2022-07-06:
i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: 20 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- 38 39 -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  

38-39 are aht10'2
20 is the mcp23017

so im not sure i will give it a try but i dont think it will work

```
python3 -m mqtt_io config.yaml
2022-07-06 01:46:27 mqtt_io.modules [DEBUG   ] Module <module 'mqtt_io.modules.gpio.mcp23017' from '/usr/local/lib/python3.9/dist-packages/mqtt_io/modules/gpio/mcp23017.py'> has all of its requirements installed already.
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/adafruit_bus_device/i2c_device.py", line 174, in __probe_for_device
    self.i2c.writeto(self.device_address, b"")
  File "/usr/local/lib/python3.9/dist-packages/busio.py", line 169, in writeto
    return self._i2c.writeto(address, buffer, stop=stop)
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 52, in writeto
    self._i2c_bus.write_bytes(address, buffer[start:end])
  File "/usr/local/lib/python3.9/dist-packages/Adafruit_PureIO/smbus.py", line 314, in write_bytes
    self._device.write(buf)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/adafruit_bus_device/i2c_device.py", line 180, in __probe_for_device
    self.i2c.readfrom_into(self.device_address, result)
  File "/usr/local/lib/python3.9/dist-packages/busio.py", line 159, in readfrom_into
    return self._i2c.readfrom_into(address, buffer, stop=stop)
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 59, in readfrom_into
    readin = self._i2c_bus.read_bytes(address, end - start)
  File "/usr/local/lib/python3.9/dist-packages/Adafruit_PureIO/smbus.py", line 181, in read_bytes
    return self._device.read(number)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/__main__.py", line 85, in <module>
    main()
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/__main__.py", line 80, in main
    mqtt_gpio.run()
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/server.py", line 1226, in run
    self._init_gpio_modules()
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/server.py", line 244, in _init_gpio_modules
    self.gpio_modules[gpio_config["name"]] = _init_module(
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/server.py", line 128, in _init_module
    return module_class(module_config)
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/modules/gpio/__init__.py", line 105, in __init__
    self.setup_module()
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/modules/gpio/mcp23017.py", line 57, in setup_module
    self.io = mcp23017.MCP23017(
  File "/usr/local/lib/python3.9/dist-packages/adafruit_mcp230xx/mcp23017.py", line 57, in __init__
    super().__init__(i2c, address)
  File "/usr/local/lib/python3.9/dist-packages/adafruit_mcp230xx/mcp23xxx.py", line 38, in __init__
    self._device = i2c_device.I2CDevice(bus_device, address)
  File "/usr/local/lib/python3.9/dist-packages/adafruit_bus_device/i2c_device.py", line 61, in __init__
    self.__probe_for_device()
  File "/usr/local/lib/python3.9/dist-packages/adafruit_bus_device/i2c_device.py", line 183, in __probe_for_device
    raise ValueError("No I2C device at address: 0x%x" % self.device_address)
ValueError: No I2C device at address: 0x22

```

so i dont get the traceback error when i use "0x20" like i do when i use an address thats not assigned.

BoW2EviL, 2022-07-06:
Okay it works perfectly now. Thank you for a wonderful project!
@bsmith76s  thanks for the advise.