flyte/mqtt-io #277: Many open Pull Requests / still any development?

mschlenstedt, 2022-07-07:
Hi,

there were a lot fixed bugs in develop comparing to 2.2.6. Still there are 11 open Pull Requests which also fixes a lot bugs. Is there any further development? Would be happy if there will be a bugfix release soon.

Can help if help is needed.

mschlenstedt, 2022-07-07:
Hey @flyte !

Thanks for the qick answer. Even if it makes me not happy :-), I can fully understand. I also have too many projects I am working on ;-)

Maybe it is worth to make one last bugfix version 2.2.7 from the develop branch including the open PR. There are some severe bugs in the raspberry module and also in the pcf8574/5 modules which makes both more or less useless at the moment:

https://github.com/flyte/mqtt-io/pull/268
19cf1e6ba3c6ad0c97f914dd888228f14fb4c4ac
1e410dc57c0a699b72ea8a6d5e38e064499a3e1b

This is maybe "quickly done" and result would be a really good and reliable piece of software. After that we (and hopefully some others, too ;-)) can focus on the trio branch.

flyte, 2022-07-07:
I've merged that and I'm just struggling with the cobwebs on the release process now...

flyte, 2022-07-07:
2.2.7 should be on PyPI now.

mschlenstedt, 2022-07-08:
@geccko  Please open a separate issue here.

geccko, 2022-07-08:
ok! Thanks!

closed 2022-11-01 by BenjiU