flyte/mqtt-io #283: All relays on after restart rpi

byniow, 2022-09-10:
After restarting rpi i have strange behavior. My config: 

```
mqtt:
  host: 10.1.0.100
  topic_prefix: rpi_io
  user: mqtt
  password: mqtt
  ha_discovery:
    enabled: yes

# GPIO
gpio_modules:
    - name: rpi
    module: raspberrypi
digital_outputs:
  - name: P1-1
    module: rpi
    pin: 5
    inverted: True

  - name: P1-2
    module: rpi
    pin: 6
    inverted: True

  - name: P1-3
    module: rpi
    pin: 13
    inverted: True
```

And after restart i get relays on. But no in the same time, but relays are on one by one after 1 second delay. Why? i use HA to control with mosquitto, and state after rpi restart is not copied from HA, but HA show all as off. 

Second - correct config for DS18b20? I try a lot options - and no success.

ondras12345, 2022-09-11:
Try adding `initial: high` (or `initial: low` if you want them to be on) and `publish_inital: true` to your relays.

This is my config for DS18B20:
```
sensor_modules:
  - name: dallas_a
    module: ds18b
    type: DS18B20
    address: 3c01e076877d

sensor_inputs:
  - name: temperature_a
    module: dallas_a
    interval: 300
    digits: 2
    ha_discovery:
      name: Temperature A
      device_class: temperature
      state_class: measurement
      unit_of_measurement: "°C"
```
You need to find the correct `address` of your device by searching in `/sys/bus/w1/devices`.

Bonus: changing resolution of DS18B20
> Writing a value between 9 and 12 to the sysfs w1_slave file will change the
> precision of the sensor for the next readings. This value is in (volatile)
> SRAM, so it is reset when the sensor gets power-cycled.
>
> ...
>
> To store the current precision configuration into EEPROM, the value 0
> has to be written to the sysfs w1_slave file. Since the EEPROM has a limited
> amount of writes (>50k), this command should be used wisely.
>
> https://www.kernel.org/doc/Documentation/w1/slaves/w1_therm

```
echo 12 | sudo tee /sys/bus/w1/devices/28-3c01e076877d/w1_slave
echo 0 | sudo tee /sys/bus/w1/devices/28-3c01e076877d/w1_slave
```

byniow, 2022-09-11:
Is there any option to restore the current switch setting in HA after RPI restart? 
There was no information in the documentation to enable 1-wire system function in raspi-config, so I was wandering.
Thank you for the guidance

ondras12345, 2022-09-11:
I don't think so, you'll need to create a custom automation. You can store the state in an `input_boolean` (which survives restart) and restore it automatically after your switch entities become available. I'm sure there will be many examples of how to do that on Home Assistant forums https://community.home-assistant.io/

byniow, 2022-09-11:
Now work. Documentation should be a little more detailed, with examples for all options. But now work with Your help. Thank You again.

closed 2022-09-11 by byniow