flyte/mqtt-io #288: mqtt_io as a service

mattb1908, 2022-10-23:
Hi Im using openhabian on a raspberry pi 4
Im trying to run mqtt_io as service, have created "runmqttio.py"

which is:
---
import os

cmd = 'python3 -m mqtt_io /home/openhabian/mqtt-io/config.MCP3008.yml'
os.system(cmd)

---

I can run this sucessfully from console using:
openhabian@openhabian:~ $ python3 /home/openhabian/runmqttio.py

but on starting "mqtt_io.service"
---

[Unit]
Description=mqtt_io
After=multi-user.target

[Service]
Type=simple
ExecStart=/usr/bin/python /home/openhabian/runmqttio.py
Restart=on-abort

[Install]
WantedBy=multi-user.target
---

i get the following:

openhabian@openhabian:~ $ sudo systemctl status mqtt_io.service
● mqtt_io.service - mqtt_io
     Loaded: loaded (/lib/systemd/system/mqtt_io.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Sun 2022-10-23 16:20:54 CEST; 3s ago
    Process: 6875 ExecStart=/usr/bin/python /home/openhabian/runmqttio.py (code=exited, status=0/SUCCESS)
   Main PID: 6875 (code=exited, status=0/SUCCESS)
        CPU: 139ms

Oct 23 16:20:54 openhabian systemd[1]: Started mqtt_io.
Oct 23 16:20:54 openhabian python[6877]: /usr/bin/python3: No module named mqtt_io
Oct 23 16:20:54 openhabian systemd[1]: mqtt_io.service: Succeeded.

although the service starts mqtt_io is not running 

I have replaced the runmqttio.py with another simple "hello_world" bit of script and that works fine!

any help much appreciated.
Matt

**Config**
```
mqtt:
  host: 192.168.0.6
  port: 1883
  user: ""
  password: ""
  topic_prefix: battery

gpio_modules:
  - name: rpi
    module: raspberrypi 

sensor_modules:
  - name: mcp3008
    module: mcp3008

digital_inputs:
  - name: test
    module: rpi
    pin: 1

sensor_inputs:
  - name: analog0
    module: mcp3008
    interval: 10
    channel: 0

  - name: analog1
    module: mcp3008
    interval: 10
    channel: 1

  - name: analog2
    module: mcp3008
    interval: 10
    channel: 2

  - name: analog3
    module: mcp3008
    interval: 10
    channel: 3

  - name: analog4
    module: mcp3008
    interval: 10
    channel: 4

  - name: analog5
    module: mcp3008
    interval: 10
    channel: 5

  - name: analog6
    module: mcp3008
    interval: 10
    channel: 6

  - name: analog7
    module: mcp3008
    interval: 10
    channel: 7

```

**Hardware**
 - Raspberry Pi,
 -

**System:**
 - OS: Openhabian]
 - Python version: Python 3.9.2
 - User you're running as: openhabian
 - Using a virtualenv?: no

ondras12345, 2022-10-23:
My guess would by that you have installed mqtt-io using pip as a regular user, but your service is running as root. (Don't try to fix this by running `pip` with `sudo`, that is not recommended.)
This is my systemd unit file (I have installed mqtt-io in a python venv in `/opt/mqttio`):
```sh
sudo tee /etc/systemd/system/mqttio.service << EOF
[Unit]
Description=MQTT IO
Requires=mosquitto.service
After=mosquitto.service

[Service]
User=homeassistant
ExecStart=/opt/mqttio/bin/python3 -m mqtt_io /etc/mqttio.yml
RestartSec=30
Restart=on-failure
SyslogIdentifier=mqttio

[Install]
WantedBy=multi-user.target
EOF
```
The important line is `User=homeassistant` (the user I want to run the program as).

mattb1908, 2022-10-23:
Brilliant, thank you so much, added line User=openhabian and its up and running.

closed 2022-10-23 by mattb1908