flyte/mqtt-io #293: mqtt-io cannot use /dev/i2c-* on Home Assistant OS 9.3 root:root/600

rlucassen2, 2022-11-04:
New to Home Assistant OS 9.3, I want to use mqtt-io for use with i2c (pcf8574). I used "HassOS I2C Configurator" to make the /dev/i2c-* available on HassOS. But as these devices have root:root and 600 as permissions, mqtt-io crashes. This is not a mqtt-io problem of course. Anyone here who knows to change these settings? I created an udev 83-i2c.rules file containing:

`SUBSYSTEM=="i2c-dev", MODE="0666"`

But no way. As the /dev/ partition is mounted ro I'm not able to change these settings. Apparently root has no rights to "mount -o remount,rw /dev" on HassOS.
A copy of the log is here: [https://tmp.xaq.nl/mqtt-io.txt](https://tmp.xaq.nl/mqtt-io.txt)

Richard.

BenjiU, 2022-11-04:
Hey, i guess it is the same problem as in #107

rlucassen2, 2022-11-04:
I just spoke to someone who said that HassOS is for people who do not know much about Linux and that instead I should run dockers on a RPi. That's what I'll do.

Richard

BenjiU, 2022-11-04:
Okay.

Please keep this comment in mind: https://github.com/flyte/mqtt-io/issues/55#issuecomment-1250077410 if you have problems with root access on docker.

Maybe the virtualenv is better on a rapsberry than running in docker. Or the setup is faster.

Thanks for the feedback.

closed 2022-11-04 by BenjiU

rlucassen2, 2022-11-04:
I have a bunch of Proxmox servers running in datacenters, but in that case I need a few km's of cable to connect the i2c bus and the serial port :-) I'll try the RPi3B, it is slow, I know, but once HA runs it should not be a problem I suppose. HassOS is terribly slow as well on a RPi3B as well, so I think I won't loose much ;-)

And thnx for your comments, I keep these in mind!

Richard

Bluscream, 2022-12-15:
How do i fix this?

BenjiU, 2022-12-15:
This must be fixed by HA. Mqtt-io can't use it, if it is not provided by the system below.