flyte/mqtt-io #294: Do not poll PCF8574/5, just use pin 13 (INT)

rlucassen2, 2022-11-05:
I have always found this a very very particular and amazing issue on the internet concerning the I/O expanders PCF8574 and PCF8575: _**nobody**_, but really _**nobody**_ uses the interrupt lines (pin 13) of the PCF I/O expander. The INT is an open collector interrupt line (use a pull-up resistor). When serializing devices (up to 16 devices using the P and AP versions), just put all pin 13 outputs together and wait until INT goes down. When this happens, one or more of the chips detected a change on one or more of its inputs!

Then, start to read the first PCF, the second etc until the INT goes up again, then you have read all new states, the rest is not necessary as there has been no change on these devices (INT isn't pulled down anymore).

I justed installed mqtt-io and I saw python3 consuming up to 5% of CPU power for just nothing. Polling a PCF8574 and PCF8575 24 hours a day is just stupid.

I have no experience in writing python scripts, so I won't be a great contributor, but if anyone can help in realizing this that would be nice!

Richard.

rlucassen2, 2022-12-07:
Hmmm, even reading 1 RPi GPIO port is done by polling AFAICS, according to the CPU power that is used :-(

Richard.

rlucassen2, 2022-12-12:
Another thing: the bits are inversed for a PCF8574, according to the datasheet:

`pinout: datasheet mqtt-io`
`pin 4: bit0 bit7`
`pin 5: bit1 bit6`
`pin 6: bit2 bit5`
`pin 7: bit3 bit4`
`pin 9: bit4 bit3`
`pin 10: bit5 bit2`
`pin 11: bit6 bit1`
`pin 12: bit7 bit0`

`/sbin/i2cset -y 1 0x20 0x01`
sets bit0 at pin 4. This mqtt-io code:

`digital_outputs:`
`  - name: "pcf8574_0x20_0"`
`    module: pcf8574`
`    pin: 0`
`    on_payload: "ON"`
`    off_payload: "OFF"`
`    initial: low`
`    ha_discovery:`
`      component: switch`
`      name: "pcf8574_0x20_0"`
`      icon: mdi:alarm-bell`

sets bit7 at pin 12.

R.

rlucassen2, 2022-12-16:
I put an INT version for Raspberry Pi 3B here: [https://xaq.nl/read-pcf/index.html](https://xaq.nl/read-pcf/index.html)

Just my 2 cts,

R.

mschlenstedt, 2023-06-23:
Interrupt for pcf8574/5 is supported by mqtt-io. Use a GPIO of the Raspberry (native support for interrupts) and connect it to the INT line of the PCF. Since the Raspberry's GPIO supports interrupts, it polls all your PCF's Inputs after detecting a change.

Needs special configuratio of the Raspberry GPIO input: https://mqtt-io.app/2.2.9-gamma/#/config/reference/digital_inputs/?id=digital_inputs-star-interrupt_for