flyte/mqtt-io #299: Sensors read but not available in MQTT

Bluscream, 2022-12-15:
- [x] I have read the [README.md](https://github.com/flyte/pi-mqtt-gpio/blob/develop/README.md) file and visited the [Documentation](https://mqtt-io.app/) to see if the information there helps.

**Describe the bug**
I want to share a DHT11 sensor with my HASS IO instance but no analog sensor gets ever broadcasted

**Expected behavior**
`home/timo/input/temperature` and `home/timo/input/humidity` topics becomes available

**Error messages and traceback**
```
Paste any error messages and Python traceback if applicable.
```

**Config**
```yaml
mqtt:
  host: home.assistant
  port: 1883
  ha_discovery:
    enabled: yes
  user: 'homeassistant'
  password: ''
  status_payload_running: available
  status_payload_stopped: unavailable
  status_payload_dead: unavailable
  topic_prefix: home/timo

gpio_modules:
  - name: rpi
    module: raspberrypi

sensor_modules:
  - name: dht11_sensor
    module: dht22
    type: DHT11
    pin: 23

sensor_inputs:
  - name: temperature
    module: dht11_sensor
    interval: 10
    digits: 4
    type: temperature
    ha_discovery:
      name: 'Timos Room Temperature'
      device_class: sensor
      state_class: measurement
      unit_of_measurement: "°C"

  - name: humidity
    module: dht11_sensor
    interval: 10
    digits: 4
    type: humidity
    ha_discovery:
      name: 'Timos Room Humidity'
      device_class: sensor
      state_class: measurement
      unit_of_measurement: "%"

digital_inputs:
  - name: automater_digital_input
    module: rpi
    pin: 0

digital_outputs:
  - name: automater_digital_output
    module: rpi
    pin: 0
```

**Hardware**
 - Platform: [e.g. Raspberry Pi, Beaglebone, Banana Pi]
 - Connected hardware:  [e.g. any external GPIO modules, sensors, streams etc.]

|  SYSTEM INFORMATION||
|----------------------|---------------------------|
|  OS type and version |     Raspbian Linux 11     |
|  Webmin version      |     2.010      |
|  Theme version       |     20.11      |
|  Package updates     |     All installed packages are up to date   |

 - Python version: 3.9.2
 - User you're running as: pi
 - Using a virtualenv?: no

<br>

## Mosquitto broker system

version | core-2022.12.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.76-v8
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4949
Installed Version | 1.28.4
Stage | running
Available Repositories | 1227
Downloaded Repositories | 4

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | January 5, 2023 at 1:00 AM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-3.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.11.2
agent_version | 1.4.1
docker_version | 20.10.19
disk_total | 57.8 GB
disk_used | 5.0 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | RPC Shutdown (2.2), File editor (5.4.2), ZeroTier One (0.15.2), Log Viewer (0.14.0), SSH & Web Terminal (13.0.0), Mosquitto broker (6.1.3), MQTT IO (0.1.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 2
views | 3
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | December 5, 2022 at 5:26 AM
-- | --
current_recorder_run | December 15, 2022 at 6:16 AM
estimated_db_size | 272.07 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

Bluscream, 2022-12-15:
I now checked the logs of my broker machine and found this error:

```
Logger: homeassistant.util.logging
Source: util/logging.py:156
First occurred: 6:18:25 AM (28 occurrences)
Last logged: 2:56:38 PM

Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'Timos Room Humidity', 'availability_topic': '/status', 'payload_available': 'available', 'payload_not_available': 'unavailable', 'device': {'manufacturer': 'MQTT IO', 'model': 'v2.2.7', 'identifiers': ['mqtt-io', 'mqtt-io-da39a3ee5e6b4b0d3255bfef95601890afd80709'], 'name': 'MQTT IO'}, 'device_class': 'sensor', 'state_class': 'measurement', 'unit_of_measurement': '%', 'unique_id': 'mqtt-io-da39a3ee5e6b4b0d3255bfef95601890afd80709_dht11_sensor_sensor_humidity', 'state_topic': '/sensor/humidity', 'expire_after': 25, 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 328, in async_discover config: DiscoveryInfoType = discovery_schema(discovery_payload) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 232, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 355, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 351, in _exec v = func(v) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable return schema(data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: expected SensorDeviceClass or one of 'date', 'duration', 'timestamp', 'apparent_power', 'aqi', 'battery', 'carbon_monoxide', 'carbon_dioxide', 'current', 'distance', 'energy', 'frequency', 'gas', 'humidity', 'illuminance', 'moisture', 'monetary', 'nitrogen_dioxide', 'nitrogen_monoxide', 'nitrous_oxide', 'ozone', 'pm1', 'pm10', 'pm25', 'power_factor', 'power', 'precipitation', 'precipitation_intensity', 'pressure', 'reactive_power', 'signal_strength', 'speed', 'sulphur_dioxide', 'temperature', 'volatile_organic_compounds', 'voltage', 'volume', 'water', 'weight', 'wind_speed' for dictionary value @ data['device_class']
Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'Timos Room Temperature', 'availability_topic': 'home/network_room/status', 'payload_available': 'available', 'payload_not_available': 'unavailable', 'device': {'manufacturer': 'MQTT IO', 'model': 'v2.2.7', 'identifiers': ['mqtt-io', 'mqtt-io-3ce486daf7b9c87344753369db50ff532c6202a1'], 'name': 'MQTT IO'}, 'device_class': 'sensor', 'state_class': 'measurement', 'unit_of_measurement': '°C', 'unique_id': 'mqtt-io-3ce486daf7b9c87344753369db50ff532c6202a1_dht11_sensor_sensor_temperature', 'state_topic': 'home/network_room/sensor/temperature', 'expire_after': 25, 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 328, in async_discover config: DiscoveryInfoType = discovery_schema(discovery_payload) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 232, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 355, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 351, in _exec v = func(v) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable return schema(data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: expected SensorDeviceClass or one of 'date', 'duration', 'timestamp', 'apparent_power', 'aqi', 'battery', 'carbon_monoxide', 'carbon_dioxide', 'current', 'distance', 'energy', 'frequency', 'gas', 'humidity', 'illuminance', 'moisture', 'monetary', 'nitrogen_dioxide', 'nitrogen_monoxide', 'nitrous_oxide', 'ozone', 'pm1', 'pm10', 'pm25', 'power_factor', 'power', 'precipitation', 'precipitation_intensity', 'pressure', 'reactive_power', 'signal_strength', 'speed', 'sulphur_dioxide', 'temperature', 'volatile_organic_compounds', 'voltage', 'volume', 'water', 'weight', 'wind_speed' for dictionary value @ data['device_class']
[... cut ...] rbon_dioxide', 'current', 'distance', 'energy', 'frequency', 'gas', 'humidity', 'illuminance', 'moisture', 'monetary', 'nitrogen_dioxide', 'nitrogen_monoxide', 'nitrous_oxide', 'ozone', 'pm1', 'pm10', 'pm25', 'power_factor', 'power', 'precipitation', 'precipitation_intensity', 'pressure', 'reactive_power', 'signal_strength', 'speed', 'sulphur_dioxide', 'temperature', 'volatile_organic_compounds', 'voltage', 'volume', 'water', 'weight', 'wind_speed' for dictionary value @ data['device_class']
```

BenjiU, 2022-12-15:
Please share the mqtt-io log files

Bluscream, 2022-12-15:
> Please share the mqtt-io log files

All the console log lines are in the post above

BenjiU, 2022-12-16:
Oh, sorry, haven't seen that on my mobile. Can't See ans error there.

I don't have HA here and no experience.

But maybe* the misstake is, that it's not home/timo/**input**/temperature but home/timo/**sensor**/temperature.
*Can't look it up right now. Just a hint

ondras12345, 2022-12-16:
> ```yaml
>     ha_discovery:
>       name: 'Timos Room Temperature'
>       device_class: sensor
>       state_class: measurement
>       unit_of_measurement: "°C"
> ```

As the Home Assistant log you posted says, this is wrong.
You should use `device_class: temperature` for this sensor and `device_class: humidity` for the other one.

closed 2024-10-17 by BenjiU