flyte/mqtt-io #306: MqttIo crashed! due to output configuration

piotruscilko, 2023-01-22:
After update to 0.2.0, add-on MQTT IO in HA, is not starting due to output configuration issue - when i delete part of the code for outputs (as below: Config of outputs), the rest is working well (mqtt io starting properly, with only sensor configuration).

**Error messages and traceback**
```
2023-01-22 08:42:21 mqtt_io.__main__ [ERROR] MqttIo crashed!
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/usr/lib/python3.10/site-packages/mqtt_io/server.py", line 1241, in run
    self._init_digital_outputs()
  File "/usr/lib/python3.10/site-packages/mqtt_io/server.py", line 453, in _init_digital_outputs
    gpio_module.setup_pin_internal(PinDirection.OUTPUT, out_conf)
  File "/usr/lib/python3.10/site-packages/mqtt_io/modules/gpio/__init__.py", line 208, in setup_pin_internal
    return self.setup_pin(
  File "/usr/lib/python3.10/site-packages/mqtt_io/modules/gpio/raspberrypi.py", line 56, in setup_pin
    self.io.setup(pin, direction, pull_up_down=pullup, initial=initial_int)
RuntimeError: No access to /dev/mem.  Try running as root!
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/usr/lib/python3.10/site-packages/mqtt_io/__main__.py", line 115, in <module>
    main()
  File "/usr/lib/python3.10/site-packages/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/usr/lib/python3.10/site-packages/mqtt_io/server.py", line 1241, in run
    self._init_digital_outputs()
  File "/usr/lib/python3.10/site-packages/mqtt_io/server.py", line 453, in _init_digital_outputs
    gpio_module.setup_pin_internal(PinDirection.OUTPUT, out_conf)
  File "/usr/lib/python3.10/site-packages/mqtt_io/modules/gpio/__init__.py", line 208, in setup_pin_internal
    return self.setup_pin(
  File "/usr/lib/python3.10/site-packages/mqtt_io/modules/gpio/raspberrypi.py", line 56, in setup_pin
    self.io.setup(pin, direction, pull_up_down=pullup, initial=initial_int)
RuntimeError: No access to /dev/mem.  Try running as root!
```

**Config of outputs**
```yaml
gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_outputs:
  - name: syrena_alarmowa
    module: raspberrypi
    pin: 14
    on_payload: "ON"
    off_payload: "OFF"
    initial: low
    ha_discovery:
      name: syrena_alarmowa
```

**Hardware**
 - Platform: Raspberry Pi

**System:**
 - OS: HASS.IO ver. 9.4
 - Home Assistant 2023.1.6
 - Add-on: MQTT IO ver. 0.2.1

BenjiU, 2023-01-22:
Hi, this ist a Problem with HASS IO, because you can't get access on /dev/mem. See #293 

Sorry

closed 2023-01-22 by BenjiU

piotruscilko, 2023-01-22:
OK, thanks for clue.

BenjiU, 2023-01-22:
No problem. What sensors are working for you? i2c?

piotruscilko, 2023-01-22:
DS18B20 connected directly to raspberry pi (default pin). At the moment it is simple configuration: 1 sensor + 1 output (relay), just prepared to further development.