flyte/mqtt-io #307: How to run when outside project directory?

codeclinic, 2023-01-22:
Hi,

How can we run this script from outside the project directory?

I really need to run this automatically at boot via crontab, but I can't figure out the correct command to get this to run unless I'm directly inside the project directory.

No matter what I try it just outputs the following:
`/python3: No module named mqtt_io`

BenjiU, 2023-01-22:
Please provide more information in your paths. What ist outside? Are you using virtualenv? Would supervisor be a solution for you?

codeclinic, 2023-01-22:
I'm not using venv or supervisor. Just cloned the project to a local directory and then run the project as described in the readme.

If I navigate to any other directory/folder I am unable to run the script.

BenjiU, 2023-07-21:
Hi @codeclinic,

if this is still relevant, please post your command lines.

closed 2024-10-17 by BenjiU