flyte/mqtt-io #318: ina219 i2c_bus_num - unknown field

renedis, 2023-06-23:
I have a ina219 sensor on i2c available on bus 10 with address 0x43. I can confirm it works with a default python script.

But with mqtt-io it doesnt.

Code:
```
root@RD-CLUST00:/srv# python3 -m mqtt_io config.yaml
2023-06-23 14:01:30 mqtt_io.__main__ [ERROR] MqttIo crashed!
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/server.py", line 1242, in run
    self._init_sensor_modules()
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/server.py", line 255, in _init_sensor_modules
    self.sensor_modules[sens_config["name"]] = _init_module(
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/server.py", line 122, in _init_module
    module_config = validate_and_normalise_config(module_config, module_schema)
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/config/__init__.py", line 105, in validate_and_normalise_config
    raise ConfigValidationFailed(
mqtt_io.exceptions.ConfigValidationFailed: Config did not validate:
i2c_bus_num:
- unknown field

Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/__main__.py", line 115, in <module>
    main()
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/server.py", line 1242, in run
    self._init_sensor_modules()
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/server.py", line 255, in _init_sensor_modules
    self.sensor_modules[sens_config["name"]] = _init_module(
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/server.py", line 122, in _init_module
    module_config = validate_and_normalise_config(module_config, module_schema)
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/config/__init__.py", line 105, in validate_and_normalise_config
    raise ConfigValidationFailed(
mqtt_io.exceptions.ConfigValidationFailed: Config did not validate:
i2c_bus_num:
- unknown field

```

yaml:
```
mqtt:
  host: X
  port: x
  user: "X"
  password: "X"
  topic_prefix: pimqttgpio/mydevice

sensor_modules:
  # An LM75 sensor attached to the I2C bus
  - name: ina219_sensor
    module: ina219
    i2c_bus_num: 10
    chip_addr: 0x43

sensor_inputs:
  # The configuration of the specific sensor value to use (LM75 only has temperature)
  - name: power
    module: ina219_sensor
```

What did I do wrong?

ondras12345, 2023-06-23:
Your config contains an extra key `i2c_bus_num`. You have probably copied it from LM75 config, but it is invalid for `ina219`.
Valid configuration for `ina219` must match this schema:
https://github.com/flyte/mqtt-io/blob/7367dd1fb34903564f27044f3f02fd8df452c04a/mqtt_io/modules/sensor/ina219.py#L31-L47

renedis, 2023-06-23:
Makes sense.. But this way it doesn't allow you to set the i2c bus number.

the Python3 script works now.. but sadly, it still does not read any values. But that's an other issue I guess.
I've already changed the script to add "i2c_bus_num" and will make a pull-request.

mschlenstedt, 2023-08-10:
I added a pull request to include i2c_bus_num. It's in the develop repo now: 2bb2f945ea47ed87f99ab4fc966d2dbbd47fc7c7