flyte/mqtt-io #320: Stream sensor (serial) not reliable

megapearl, 2023-07-12:
First of all thanks for your time and great code.

- [ ] I have read the [README.md](https://github.com/flyte/pi-mqtt-gpio/blob/develop/README.md) file and visited the [Documentation](https://mqtt-io.app/) to see if the information there helps.

**Describe the bug**
The string that comes from the serial port is sometimes incomplete.

**Expected behavior**
A complete string

**Error messages and traceback**
```
No errors
```

**Config**
```yaml
mqtt:
  host: shed-mqtt.flissinger.com
  port: 1883
  user: *redacted*
  password: *redacted*
  topic_prefix: home-domotica
  client_id: cupboard-domotica
  status_topic: cupboard_domotica/status
  reconnect_delay: 1

gpio_modules:
  - name: mcp23017_binary_sensor
    module: mcp23017
    chip_addr: 0x20
  - name: mcp23017_switch
    module: mcp23017
    chip_addr: 0x21

digital_inputs:
  - name: hallway_frontdoor
    module: mcp23017_binary_sensor
    pin: 14
    inverted: false
    pullup: true
    retain: true
    ha_discovery:
      component: binary_sensor
      name: Hallway Frontdoor

digital_outputs:
  - name: frontdoor_lock
    module: mcp23017_switch
    pin: 0
    inverted: true
    retain: true
    initial: high
    publish_initial: true
    ha_discovery:
      component: switch
      name: Frontdoor Lock
      device_class: lock

stream_modules:
  - name: cupboard_alarm_stream
    module: serial
    device: /dev/ttyUSB1
    baud: 19200
    bytesize: 8
    parity: none
    stopbits: 1
    cleanup: true
    retain: true
    read: true
    write: false
    read_interval: 0.25

```

**Hardware**
 - Platform: Raspberry Pi 4
 - Connected hardware:  mcp23017, usb to serial dongle (startech)

**System:**
 - OS: Linux cupboard-domotica 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
 - Python version: Python 3.9.2
 - User you're running as: gpio2mqtt
 - Using a virtualenv?: yes

**Additional context**
I hooked up an old alarm system which has a serial "printer" port, that "prints" strings to the serial port.
When someone uses a badge or opens a door, the alarm system output it to the serial port.

The strings that mqtt-io reads are incomplete, They should be something like:
"4560 21:23:37 10/07/23 Access granted on Door 3 for user 2 Donald"

In 5 of 10 cases this is working as expected, but most of the time the string received is a part of it like:
"es granted on Door 3 for user 2 Donald"

I tried different versions of mqtt-io.

```
cd /home/gpio2mqtt
source bin/activate
#pip install --upgrade mqtt-io
#pip install --upgrade git+https://github.com/flyte/mqtt-io.git@feature/trio
#pip install --upgrade git+https://github.com/flyte/mqtt-io.git@develop
pip install --upgrade git+https://github.com/flyte/mqtt-io.git@2.2.9
```

I tried using different read_interval for the stream_modules, from 0.1 to 2.0

I read through:
https://mqtt-io.app/2.2.9/#/config/reference/stream_modules/
https://github.com/flyte/mqtt-io/blob/develop/mqtt_io/modules/stream/serial.py
to find other options like buffering but could not find it.

I tried cleanup: false and cleanup: true
It doesn't make any difference, what else can I try?

What could be the problem? it seems like a buffer issue to me, but I don't know for sure.

Best Regards,
Donald.

BenjiU, 2024-10-17:
Hi @megapearl,

is this issue still present on your setup? Maybe it is fixed after some improvements?

Thanks
Benni

megapearl, 2024-10-17:
Yes, the issue is still present