flyte/mqtt-io #322: Version 2.2.9 is not available on PyPI

ondras12345, 2023-07-14:
It has been 2 days since the release was published on Github, so I assume it would already be on PyPI if everything was working right.
https://pypi.org/project/mqtt-io/#history

Apparently the "Publish to PyPI" action was skipped: https://github.com/flyte/mqtt-io/actions/runs/5530576216

lloydbayley, 2023-07-17:
Any joy here? Would like to upgrade!

BenjiU, 2023-07-18:
Hey, it's sonerhing with the tag variable in the pipeline. I'll try to fix it today. Next three days I'm going to be on a workshop and won't be available...

BenjiU, 2023-07-18:
@ondras12345 @lloydbayley
2.2.9 should be availabe on pypi.org now. May you please have a look at it and verify if it is working?

Thanks!
Benni

lloydbayley, 2023-07-19:
> @ondras12345 @lloydbayley 2.2.9 should be availabe on pypi.org now. May you please have a look at it and verify if it is working?
> 
> Thanks! Benni

It is indeed there! Thank you very much!
Always nice when you can say "It VERKS!" :)

closed 2023-07-19 by BenjiU