flyte/mqtt-io #324: Pinned PyYAML version incompatible with latest Cython 3.0.0

frenck, 2023-07-19:
- [x] I have read the [README.md](https://github.com/flyte/pi-mqtt-gpio/blob/develop/README.md) file and visited the [Documentation](https://mqtt-io.app/) to see if the information there helps.

**Describe the bug**

`pip install` with everything latest greatest in a clean environment will pull in cython 3.0.0, which will not work with the PyYAML version pinned by this project.

```txt
#9 24.58 Collecting PyYAML<6.0,>=5.4
#9 24.60   Downloading PyYAML-5.4.1.tar.gz (175 kB)
#9 24.60      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 175.1/175.1 kB 200.7 MB/s eta 0:00:00
#9 24.76   Installing build dependencies: started
#9 29.20   Installing build dependencies: finished with status 'done'
#9 29.21   Getting requirements to build wheel: started
#9 29.45   Getting requirements to build wheel: finished with status 'error'
#9 29.46   error: subprocess-exited-with-error
#9 29.46   
#9 29.46   × Getting requirements to build wheel did not run successfully.
#9 29.46   │ exit code: 1
#9 29.46   ╰─> [68 lines of output]
#9 29.46       /tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/config/setupcfg.py:293: _DeprecatedConfig: Deprecated config in `setup.cfg`
#9 29.46       !!
#9 29.46       
#9 29.46               ********************************************************************************
#9 29.46               The license_file parameter is deprecated, use license_files instead.
#9 29.46       
#9 29.46               By 2023-Oct-30, you need to update your project and remove deprecated calls
#9 29.46               or your builds will no longer be supported.
#9 29.46       
#9 29.46               See https://setuptools.pypa.io/en/latest/userguide/declarative_config.html for details.
#9 29.46               ********************************************************************************
#9 29.46       
#9 29.46       !!
#9 29.46         parsed = self.parsers.get(option_name, lambda x: x)(value)
#9 29.46       running egg_info
#9 29.46       writing lib3/PyYAML.egg-info/PKG-INFO
#9 29.46       writing dependency_links to lib3/PyYAML.egg-info/dependency_links.txt
#9 29.46       writing top-level names to lib3/PyYAML.egg-info/top_level.txt
#9 29.46       Traceback (most recent call last):
#9 29.46         File "/usr/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
#9 29.46           main()
#9 29.46         File "/usr/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
#9 29.46           json_out['return_val'] = hook(**hook_input['kwargs'])
#9 29.46                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
#9 29.46         File "/usr/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
#9 29.46           return hook(config_settings)
#9 29.46                  ^^^^^^^^^^^^^^^^^^^^^
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 341, in get_requires_for_build_wheel
#9 29.46           return self._get_build_requires(config_settings, requirements=['wheel'])
#9 29.46                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 323, in _get_build_requires
#9 29.46           self.run_setup()
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 338, in run_setup
#9 29.46           exec(code, locals())
#9 29.46         File "<string>", line 271, in <module>
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/__init__.py", line 107, in setup
#9 29.46           return distutils.core.setup(**attrs)
#9 29.46                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 185, in setup
#9 29.46           return run_commands(dist)
#9 29.46                  ^^^^^^^^^^^^^^^^^^
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
#9 29.46           dist.run_commands()
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
#9 29.46           self.run_command(cmd)
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/dist.py", line 1234, in run_command
#9 29.46           super().run_command(command)
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
#9 29.46           cmd_obj.run()
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/command/egg_info.py", line 314, in run
#9 29.46           self.find_sources()
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/command/egg_info.py", line [322](https://github.com/hassio-addons/addon-mqtt-io/actions/runs/5591727199/jobs/10223217322#step:10:326), in find_sources
#9 29.46           mm.run()
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/command/egg_info.py", line 551, in run
#9 29.46           self.add_defaults()
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/command/egg_info.py", line 589, in add_defaults
#9 29.46           sdist.add_defaults(self)
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/command/sdist.py", line 104, in add_defaults
#9 29.46           super().add_defaults()
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/_distutils/command/sdist.py", line 251, in add_defaults
#9 29.46           self._add_defaults_ext()
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/_distutils/command/sdist.py", line 336, in _add_defaults_ext
#9 29.46           self.filelist.extend(build_ext.get_source_files())
#9 29.46                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
#9 29.46         File "<string>", line 201, in get_source_files
#9 29.46         File "/tmp/pip-build-env-7f1pbprl/overlay/lib/python3.11/site-packages/setuptools/_distutils/cmd.py", line 107, in __getattr__
#9 29.46           raise AttributeError(attr)
#9 29.46       AttributeError: cython_sources
#9 29.46       [end of output]
#9 29.46   
#9 29.46   note: This error originates from a subprocess, and is likely not a problem with pip.
#9 29.47 error: subprocess-exited-with-error
#9 29.47 
#9 29.47 × Getting requirements to build wheel did not run successfully.
#9 29.47 │ exit code: 1
#9 29.47 ╰─> See above for output.
#9 29.47
```

**Expected behavior**

To be able to install.

**Error messages and traceback**
```
See above
```

**Config**
```yaml
Not relevant.
```

**Hardware**
 - Platform: Any
 - Connected hardware: Not relevant

**System:**
 - OS: Any
 - Python version: Python 3.11
 - User you're running as: Irrelevant
 - Using a virtualenv?: yes

**Additional context**

This issue has been solved in PyYAML 6.0.1

Changelog: <https://github.com/yaml/pyyaml/blob/6.0.1/CHANGES#L7>
Ref: https://github.com/yaml/pyyaml/compare/6.0...6.0.1

**Possible solutions**

- Update PyYAML to be at least 6.0.1 or newer
- If that is not possible, pinning cython dep to < 3.0 in this project might be a workaround

frenck, 2023-07-21:
Can we reopen this issue @BenjiU?

The issue has been marked solve, but is actually not correctly resolved. The current dependency states PyYAML 6.0.0 is working, but it is not.

../Frenck

BenjiU, 2023-07-21:
Yes, of course. What does not work?

frenck, 2023-07-21:
In #325, you've pinned PyYAML to `^6.0`, meaning `6.0.0` would be a valid version to use.

See: <https://github.com/flyte/mqtt-io/blob/0317539150e15e6ed5fd4c11e0b9d88294e7970a/pyproject.toml#L12>

However, `6.0.0` has the exact same problem as I have reported in this GitHub issue. As pointed out in the possible solutions section in the issue report, you need to ping `6.0.1` as a minimum to resolve the issue, or, pin Cython (which is more of a workaround).

../Frenck

BenjiU, 2023-07-21:
Yeah, okay. But ist this a practical problem? Because IT pulls the 6.0.1 version:
`#42` 260.3   Created wheel for pyyaml: filename=PyYAML-6.0.1-cp38-cp38-linux_armv7l.whl size=542410 `

frenck, 2023-07-21:
Yes, that is a practical problem and not correct.

closed 2023-07-22 by BenjiU

frenck, 2023-07-22:
Thanks, @BenjiU 👍 

../Frenck

commit d7e07f5f (2023-07-22): pin pyyaml to v6.0.1 (#326)

* fixes #324