flyte/mqtt-io #334: pip fails on pyyaml, poetry fails on requests.utils on a raspberry pi zero w

novski, 2023-10-23:
- [x] I have read the [README.md](https://github.com/flyte/pi-mqtt-gpio/blob/develop/README.md) file and visited the [Documentation](https://mqtt-io.app/) to see if the information there helps.

**Describe the bug**
Installation on raspberry pi zero w fails with pip and poetry.

**Expected behavior**
pip should install mqtt-io.

**Error messages and traceback**
```
~/mqtt-io $ pip install mqtt-io
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting mqtt-io
  Downloading https://www.piwheels.org/simple/mqtt-io/mqtt_io-2.2.9-py3-none-any.whl (81 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 81.5/81.5 kB 409.6 kB/s eta 0:00:00
Collecting Cerberus<2.0.0,>=1.3.2
  Downloading https://www.piwheels.org/simple/cerberus/Cerberus-1.3.5-py3-none-any.whl (30 kB)
Collecting PyYAML<6.0,>=5.4
  Downloading PyYAML-5.4.1.tar.gz (175 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 175.1/175.1 kB 358.8 kB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [68 lines of output]
      /tmp/pip-build-env-ecwtibsm/overlay/lib/python3.11/site-packages/setuptools/config/setupcfg.py:293: _DeprecatedConfig: Deprecated config in `setup.cfg`
      !!
      
              ********************************************************************************
              The license_file parameter is deprecated, use license_files instead.
      
              By 2023-Oct-30, you need to update your project and remove deprecated calls
              or your builds will no longer be supported.
      
              See https://setuptools.pypa.io/en/latest/userguide/declarative_config.html for details.
              ********************************************************************************
      
      !!
        parsed = self.parsers.get(option_name, lambda x: x)(value)
      running egg_info
      writing lib3/PyYAML.egg-info/PKG-INFO
      writing dependency_links to lib3/PyYAML.egg-info/dependency_links.txt
      writing top-level names to lib3/PyYAML.egg-info/top_level.txt
      Traceback (most recent call last):
        File "/home/novski/mqtt-io/.venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/novski/mqtt-io/.venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/home/novski/mqtt-io/.venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-ecwtibsm/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 355, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-ecwtibsm/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 325, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-ecwtibsm/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 341, in run_setup
          exec(code, locals())
        File "<string>", line 271, in <module>
        File "/tmp/pip-build-env-ecwtibsm/overlay/lib/python3.11/site-packages/setuptools/__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-ecwtibsm/overlay/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 185, in setup
          return run_commands(dist)
                 ^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-ecwtibsm/overlay/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
          dist.run_commands()
[... 539 lines cut ...]
  File "/home/novski/mqtt-io/.venv/lib/python3.11/site-packages/poetry/utils/helpers.py", line 17, in <module>
    from requests.utils import atomic_open
ImportError: cannot import name 'atomic_open' from 'requests.utils' (/home/novski/mqtt-io/.venv/lib/python3.11/site-packages/requests/utils.py)
```
where a reinstallation of `pip install utils` does not help and throws the exact same as above again.

Here i got stuck... 

**Config**
```yaml
not yet done.
```

**Hardware**
 - Platform: raspberry pi zero w
 - Connected hardware: none
**System:**
 - OS: Raspbian lite 32bit
 - Python version: 3.11.2
 - Pip version: 23.0.1
 - User you're running as: my_personal_user
 - Using a virtualenv?: yes

**Additional context**
Add any other context about the problem here.

novski, 2023-10-23:
found some more hints: 
poetry 1.6 has a bug that didn't got fixed with its pullrequest to 1.6.1: https://github.com/python-poetry/poetry/pull/8336
so to get over the atomic_open import fail which is because `requests <2.26 I reinstalled it with: `pip install --force-reinstall -v "requests==2.31"`
and got this funny warning:
```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
poetry 1.6.1 requires platformdirs<4.0.0,>=3.0.0, but you have platformdirs 2.4.0 which is incompatible.
```
then did: `pip install --force-reinstall -v "platformdirs==3.11.0"` & `pip install --force-reinstall -v "filelock==3.12.4"` as it showed the same error...
I don't know if the pip installation has something to do with the poetry installation, but I leave it up to you to take that in to account...

terikin, 2023-10-24:
I'm having the same problem installing on a Raspberry Pi 4 B with the latest Raspberry Pi OS (64 bit).  I've only tried installing via `pip3 install mqtt-io`

greenembrace, 2023-10-24:
I'm having a similar issue
RPi 3B+
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Python 3.11.4
pip 23.3.1

adargu, 2023-10-25:
Also ended here with the same Problem

Raspberry Pi 2 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
Python 3.11.2
pip 23.3.1

terikin, 2023-10-26:
So I believe I found a workaround.  Something to do with PyYAML not having a maximum working version of Cython specified, and Cython 3.0 broke something.

Thanks to the "Inline Constraint" on [this page](https://github.com/yaml/pyyaml/issues/736).

Follow that link for more details, but here's the short version of what I did before calling `pip install mqtt-io`:

```bash
# create a constraint file that limits the Cython version to one that should work
echo 'Cython < 3.0' > /tmp/constraint.txt

# install PyYAML itself (or other packages that need it); any package requiring Cython will be constrained to `Cython < 3.0`
PIP_CONSTRAINT=/tmp/constraint.txt pip install 'PyYAML==5.4.1'

# install mqtt-io, now that we have a working version of pyyaml
pip install mqtt-io
```

novski, 2023-10-27:
I orderd an Pi Zero 2 W to test if it does not fail on more recent architecture and OS. So i tested it with 64bit Raspberrypi OS:
```
$dmesg
...
[    0.000000] Linux version 6.1.0-rpi4-rpi-v8 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05)
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi Zero 2 W Rev 1.0
...
```
but that fails as well with the same reasons.
Just to let anybody know, that you don't have to order different hardware...

mschlenstedt, 2023-12-31:
Workaround works for me, too!

I am on DietPi with Debian Bookworm (Python 3.11).

mschlenstedt, 2024-04-06:
THis seems to be fixed in latest mqtt-io 2.3.0?