flyte/mqtt-io #338: KeyError when trying to run on Orange Pi Zero 2W, Ubuntu 22.04

flensimdotcore, 2023-11-23:
**Action description**
I've set a digital outpu pin 38 and it threw a KeyError: 38 error.

**Previous attempts**
I've tried a few different pin numbers and all of them threw this error.

**Error messages and traceback**
```
orangepi@orangepizero2w:~/Desktop$ python3 -m mqtt_io /home/orangepi/Desktop/config.yml
/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/modules/gpio/orangepi.py:63: UserWarning: Pull up/down setting are not (yet) fully supported, continuing anyway. Use GPIO.setwarnings(False) to disable warnings.
  self.io.setup(pin, direction, pull_up_down=pullup, initial=initial_int)
2023-11-23 18:29:48 mqtt_io.__main__ [ERROR] MqttIo crashed!
Traceback (most recent call last):
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/server.py", line 1271, in run
    self._init_digital_outputs()
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/server.py", line 453, in _init_digital_outputs
    gpio_module.setup_pin_internal(PinDirection.OUTPUT, out_conf)
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/modules/gpio/__init__.py", line 208, in setup_pin_internal
    return self.setup_pin(
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/modules/gpio/orangepi.py", line 63, in setup_pin
    self.io.setup(pin, direction, pull_up_down=pullup, initial=initial_int)
  File "/home/orangepi/.local/lib/python3.10/site-packages/OPi/GPIO.py", line 470, in setup
    pin = get_gpio_pin(_mode, channel)
  File "/home/orangepi/.local/lib/python3.10/site-packages/OPi/pin_mappings.py", line 80, in get_gpio_pin
    return _pin_map[mode][channel]
KeyError: 38
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/__main__.py", line 115, in <module>
    main()
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/server.py", line 1271, in run
    self._init_digital_outputs()
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/server.py", line 453, in _init_digital_outputs
    gpio_module.setup_pin_internal(PinDirection.OUTPUT, out_conf)
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/modules/gpio/__init__.py", line 208, in setup_pin_internal
    return self.setup_pin(
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/modules/gpio/orangepi.py", line 63, in setup_pin
    self.io.setup(pin, direction, pull_up_down=pullup, initial=initial_int)
  File "/home/orangepi/.local/lib/python3.10/site-packages/OPi/GPIO.py", line 470, in setup
    pin = get_gpio_pin(_mode, channel)
  File "/home/orangepi/.local/lib/python3.10/site-packages/OPi/pin_mappings.py", line 80, in get_gpio_pin
    return _pin_map[mode][channel]
KeyError: 38
```

**Config**
```yaml
mqtt:
  host: test.mosquitto.org
  port: 1883
  user: ""
  password: ""
  topic_prefix: home/livingroom/sockets
  
gpio_modules:
  - name: orangepi
    module: orangepi
    mode: board

digital_outputs:
  - name: make_small_cup
    module: orangepi
    pin: 38
    on_payload: "ON"
    off_payload: "OFF"
    initial: low
```

**Hardware**
 - Platform: Orange Pi (Orange Pi Zero 2W)
 - Connected hardware:  none

**System:**
 - OS: Ubuntu 22.04
 - Python version: Python 3.10.12
 - User you're running as: pi
 - Using a virtualenv?: no

flensimdotcore, 2023-11-23:
I tried to set a 12 pin, and now I'm getting a new error:
```
orangepi@orangepizero2w:~/Desktop$ python3 -m mqtt_io /home/orangepi/Desktop/config.yml
/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/modules/gpio/orangepi.py:63: UserWarning: Pull up/down setting are not (yet) fully supported, continuing anyway. Use GPIO.setwarnings(False) to disable warnings.
  self.io.setup(pin, direction, pull_up_down=pullup, initial=initial_int)
2023-11-23 18:53:11 mqtt_io.__main__ [ERROR] MqttIo crashed!
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/server.py", line 1271, in run
    self._init_digital_outputs()
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/server.py", line 453, in _init_digital_outputs
    gpio_module.setup_pin_internal(PinDirection.OUTPUT, out_conf)
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/modules/gpio/__init__.py", line 208, in setup_pin_internal
    return self.setup_pin(
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/modules/gpio/orangepi.py", line 63, in setup_pin
    self.io.setup(pin, direction, pull_up_down=pullup, initial=initial_int)
  File "/home/orangepi/.local/lib/python3.10/site-packages/OPi/GPIO.py", line 480, in setup
    raise e
  File "/home/orangepi/.local/lib/python3.10/site-packages/OPi/GPIO.py", line 472, in setup
    sysfs.export(pin)
  File "/home/orangepi/.local/lib/python3.10/site-packages/OPi/sysfs.py", line 37, in export
    with open(path, "w") as fp:
OSError: [Errno 22] Invalid argument
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/__main__.py", line 115, in <module>
    main()
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/server.py", line 1271, in run
    self._init_digital_outputs()
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/server.py", line 453, in _init_digital_outputs
    gpio_module.setup_pin_internal(PinDirection.OUTPUT, out_conf)
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/modules/gpio/__init__.py", line 208, in setup_pin_internal
    return self.setup_pin(
  File "/home/orangepi/.local/lib/python3.10/site-packages/mqtt_io/modules/gpio/orangepi.py", line 63, in setup_pin
    self.io.setup(pin, direction, pull_up_down=pullup, initial=initial_int)
  File "/home/orangepi/.local/lib/python3.10/site-packages/OPi/GPIO.py", line 480, in setup
    raise e
  File "/home/orangepi/.local/lib/python3.10/site-packages/OPi/GPIO.py", line 472, in setup
    sysfs.export(pin)
  File "/home/orangepi/.local/lib/python3.10/site-packages/OPi/sysfs.py", line 37, in export
    with open(path, "w") as fp:
OSError: [Errno 22] Invalid argument
```

BenjiU, 2024-02-26:
Hi @flensimdotcore,

have you continued investigating this issue? Looks like something is wrong with the "path".
Are you able to test it with suggestions from here: https://stackoverflow.com/questions/25584124/oserror-errno-22-invalid-argument-when-use-open-in-python

Thanks,
Benni