flyte/mqtt-io #340: Raspberry pi 5 support?

deanfourie1, 2024-01-06:
Hi I am assuming that this will not support the new Raspberry PI 5 hardware?

Thanks

BenjiU, 2024-02-26:
Hi @deanfourie1,

I hope it does... Have you had the possibility to try it?

Thanks!
Benni

JanFellner, 2024-08-17:
Just humbling over this topic. I am using https://wiki.loxberry.de/plugins/multi_io/start which is based on mqtt-io and it is not working with raspi 5 at the moment...

`2024-08-17 11:55:33,329 mqtt_io.__main__ (ERROR): MqttIo crashed!
Traceback (most recent call last):
File "/usr/local/lib/python3.11/dist-packages/mqtt_io/__main__.py", line 107, in main
mqtt_gpio.run()
File "/usr/local/lib/python3.11/dist-packages/mqtt_io/server.py", line 1448, in run
self._init_digital_inputs()
File "/usr/local/lib/python3.11/dist-packages/mqtt_io/server.py", line 420, in _init_digital_inputs
gpio_module.setup_pin_internal(PinDirection.INPUT, in_conf)
File "/usr/local/lib/python3.11/dist-packages/mqtt_io/modules/gpio/__init__.py", line 208, in setup_pin_internal
return self.setup_pin(
^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/dist-packages/mqtt_io/modules/gpio/raspberrypi.py", line 56, in setup_pin
self.io.setup(pin, direction, pull_up_down=pullup, initial=initial_int)
RuntimeError: Cannot determine SOC peripheral base address
Traceback (most recent call last):
File " ", line 198, in _run_module_as_main
File " ", line 88, in _run_code
File "/usr/local/lib/python3.11/dist-packages/mqtt_io/__main__.py", line 115, in
main()
File "/usr/local/lib/python3.11/dist-packages/mqtt_io/__main__.py", line 107, in main
mqtt_gpio.run()
File "/usr/local/lib/python3.11/dist-packages/mqtt_io/server.py", line 1448, in run
self._init_digital_inputs()
File "/usr/local/lib/python3.11/dist-packages/mqtt_io/server.py", line 420, in _init_digital_inputs
gpio_module.setup_pin_internal(PinDirection.INPUT, in_conf)
File "/usr/local/lib/python3.11/dist-packages/mqtt_io/modules/gpio/__init__.py", line 208, in setup_pin_internal
return self.setup_pin(
^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/dist-packages/mqtt_io/modules/gpio/raspberrypi.py", line 56, in setup_pin
self.io.setup(pin, direction, pull_up_down=pullup, initial=initial_int)
RuntimeError: Cannot determine SOC peripheral base address`

ondras12345, 2024-12-06:
I was able to migrate my mqttio from RPi4 to RPi5 by switching to gpiozero:
```yaml
gpio_modules:
  - name: rpi
    module: gpiozero
```
And setting this environment variable:`GPIOZERO_PIN_FACTORY=lgpio`.
For some reason, I also had to invert all my `digital_inputs`.

khseal, 2025-07-09:
[gpiozero.txt](https://github.com/user-attachments/files/21143596/gpiozero.txt)
Fix gpiozero.py for pi5. Fix digital_outputs and initial.