flyte/mqtt-io #348: Issues with paho-mqtt 2.0

jon-daemon, 2024-02-11:
**Describe the bug**
I couldn't start mqttio and I was getting the errors below. After finding that paho-mqtt 2.0 was installed and it was released only two days ago, I tried the previous paho-mqtt==1.6.1 and the error was gone. I think you should add a requirement of paho-mqtt < 2

**Error messages and traceback**
```
/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/mqtt_io/modules/gpio/orangepi.py:63: UserWarning: Pull up/down setting are not (yet) fully supported, continuing anyway. Use GPIO.setwarnings(False) to disable warnings.
  self.io.setup(pin, direction, pull_up_down=pullup, initial=initial_int)
Exception ignored in: <function Client.__del__ at 0xb51c3528>
Traceback (most recent call last):
  File "/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
           ^^^^^^^^^^
AttributeError: 'Client' object has no attribute '_sock'
2024-02-12 01:00:06 mqtt_io.__main__ [ERROR] MqttIo crashed!
Traceback (most recent call last):
  File "/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/mqtt_io/server.py", line 1280, in run
    self.loop.run_until_complete(self._main_task)
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/mqtt_io/server.py", line 1199, in _main_loop
    await self._connect_mqtt()
  File "/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/mqtt_io/server.py", line 591, in _connect_mqtt
    self.mqtt = AbstractMQTTClient.get_implementation(config["client_module"])(
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/mqtt_io/mqtt/asyncio_mqtt.py", line 63, in __init__
    self._client = Client(
                   ^^^^^^^
  File "/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/asyncio_mqtt/client.py", line 34, in __init__
    self._client = mqtt.Client(client_id=client_id, protocol=protocol, clean_session=clean_session, transport=transport)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: Client.__init__() missing 1 required positional argument: 'callback_api_version'
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/mqtt_io/__main__.py", line 115, in <module>
    main()
  File "/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/mqtt_io/server.py", line 1280, in run
    self.loop.run_until_complete(self._main_task)
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/mqtt_io/server.py", line 1199, in _main_loop
    await self._connect_mqtt()
  File "/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/mqtt_io/server.py", line 591, in _connect_mqtt
    self.mqtt = AbstractMQTTClient.get_implementation(config["client_module"])(
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/mqtt_io/mqtt/asyncio_mqtt.py", line 63, in __init__
    self._client = Client(
                   ^^^^^^^
  File "/home/pi/mqttio/mqttio-venv/lib/python3.11/site-packages/asyncio_mqtt/client.py", line 34, in __init__
    self._client = mqtt.Client(client_id=client_id, protocol=protocol, clean_session=clean_session, transport=transport)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: Client.__init__() missing 1 required positional argument: 'callback_api_version'
```

**Hardware**
 - Platform: Orange Pi Zero

**System:**
 - OS: Debian 12
 - Python version: 3.11
 - User you're running as: pi
 - Using a virtualenv?: yes

linesquarecube, 2024-03-02:
Thanks for sharing this information @jon-daemon!
Saved me a lot of time figuring out why me fresh install was not working :smiley:

henryptung, 2024-03-16:
Note: related to this, but I think asyncio-mqtt is a defunct project name. It was renamed in 1.0 to [aiomqtt](https://github.com/sbtinstruments/aiomqtt/releases/tag/v1.0.0), so if someone wants to do the bump, that should be bumped too.

n8murillo, 2024-03-23:
Sorry but am I missing the workaround?

BenjiU, 2024-04-09:
As @henryptung mentioned, i tried to update paho_mqtt to version 2.0.0, but this is incompatible with asyncio-mqtt. And updating asyncio-mqtt to aiomqtt pins paho_mqtt to version 1.6.0

I'll work on #364. But I have to fix a lot of lint warnings (due to updated module...).

Feel free to have a look or contribute in #364 :-)

markferry, 2024-04-28:
aiomqtt have now released a paho-mqtt v2 version: https://github.com/sbtinstruments/aiomqtt/releases/tag/v2.1.0

BenjiU, 2024-07-30:
fixed

closed 2024-07-30 by BenjiU