flyte/mqtt-io #355: New version not working

Bluscream, 2024-03-02:
raspi docker a64 user here. im on mobile so excuse the less details here.

the new update seems to have broken the docker container. earlier today i noticed my mqtt-io container wasnt running anymore and when i checked the logs it said that setcap wasnt found a lot. When i manually removed the setcap from that start cmd it said that gcc failed when installing rpi.gpio

BenjiU, 2024-03-06:
Hi @Bluscream,

have you been on a normal PC in the while? :-D Can you provide more information, please?
I don't have the setup here for reproduction.

Thanks
Benni

Bluscream, 2024-03-06:
docker-compose:
```yaml
version: '3'
services:
  mqtt-io:
    privileged: true
    user: root
    image: flyte/mqtt-io
    ports:
     - "443:443"
     - "1883:1883"
     - "8883:8883"
     - "14567:14567"
    volumes:
      - type: bind
        source: /portainer/Files/AppData/Config/mqtt-io/config.yml
        target: /config.yml
        read_only: true
      - /dev/i2c-0:/dev/i2c-0
      - /dev/mem:/dev/mem
      - /dev/gpiomem:/dev/gpiomem
    devices:
      - /dev/i2c-0
      - /dev/mem
      - /dev/gpiomem
    command: >
      bash -c "setcap 'cap_sys_rawio+eip' $(readlink -f venv/bin/python) && venv/bin/python -m mqtt_io /config.yml"
    network_mode: host
    ```
    
    logs:
    ![](https://i.imgur.com/IjNGSd3.png)

Bluscream, 2024-03-06:
New command in docker-compose without setcap:
`'bash' '-c' 'venv/bin/python -m mqtt_io /config.yml'`

Log without setcap:
```log
Collecting RPi.GPIO
  Downloading RPi.GPIO-0.7.1.tar.gz (29 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Building wheels for collected packages: RPi.GPIO
  Building wheel for RPi.GPIO (setup.py): started
  Building wheel for RPi.GPIO (setup.py): finished with status 'error'
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [16 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-armv7l-3.8
      creating build/lib.linux-armv7l-3.8/RPi
      copying RPi/__init__.py -> build/lib.linux-armv7l-3.8/RPi
      creating build/lib.linux-armv7l-3.8/RPi/GPIO
      copying RPi/GPIO/__init__.py -> build/lib.linux-armv7l-3.8/RPi/GPIO
      running build_ext
      building 'RPi._GPIO' extension
      creating build/temp.linux-armv7l-3.8
      creating build/temp.linux-armv7l-3.8/source
      gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/mqtt_io/venv/include -I/usr/local/include/python3.8 -c source/py_gpio.c -o build/temp.linux-armv7l-3.8/source/py_gpio.o
      unable to execute 'gcc': No such file or directory
      error: command 'gcc' failed with exit status 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for RPi.GPIO
  Running setup.py clean for RPi.GPIO
Failed to build RPi.GPIO
ERROR: Could not build wheels for RPi.GPIO, which is required to install pyproject.toml-based projects
2024-03-06 22:17:41 mqtt_io.__main__ [ERROR] MqttIo crashed!
Traceback (most recent call last):
  File "/home/mqtt_io/mqtt_io/modules/__init__.py", line 51, in install_missing_module_requirements
    install_missing_requirements(pkgs_required)
  File "/home/mqtt_io/mqtt_io/modules/__init__.py", line 22, in install_missing_requirements
    check_call([sys.executable, "-m", "pip", "install"] + pkgs_required)
  File "/usr/local/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/mqtt_io/venv/bin/python', '-m', 'pip', 'install', 'RPi.GPIO']' returned non-zero exit status 1.
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/home/mqtt_io/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/home/mqtt_io/mqtt_io/server.py", line 1447, in run
    self._init_gpio_modules()
  File "/home/mqtt_io/mqtt_io/server.py", line 273, in _init_gpio_modules
    self.gpio_modules[gpio_config["name"]] = _init_module(
  File "/home/mqtt_io/mqtt_io/server.py", line 127, in _init_module
    install_missing_module_requirements(module)
  File "/home/mqtt_io/mqtt_io/modules/__init__.py", line 53, in install_missing_module_requirements
    raise CannotInstallModuleRequirements(
mqtt_io.exceptions.CannotInstallModuleRequirements: Unable to install packages for module <module 'mqtt_io.modules.gpio.raspberrypi' from '/home/mqtt_io/mqtt_io/modules/gpio/raspberrypi.py'> (['RPi.GPIO']): Command '['/home/mqtt_io/venv/bin/python', '-m', 'pip', 'install', 'RPi.GPIO']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/home/mqtt_io/mqtt_io/modules/__init__.py", line 51, in install_missing_module_requirements
    install_missing_requirements(pkgs_required)
  File "/home/mqtt_io/mqtt_io/modules/__init__.py", line 22, in install_missing_requirements
    check_call([sys.executable, "-m", "pip", "install"] + pkgs_required)
  File "/usr/local/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/mqtt_io/venv/bin/python', '-m', 'pip', 'install', 'RPi.GPIO']' returned non-zero exit status 1.
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/mqtt_io/mqtt_io/__main__.py", line 115, in <module>
    main()
  File "/home/mqtt_io/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/home/mqtt_io/mqtt_io/server.py", line 1447, in run
    self._init_gpio_modules()
  File "/home/mqtt_io/mqtt_io/server.py", line 273, in _init_gpio_modules
    self.gpio_modules[gpio_config["name"]] = _init_module(
  File "/home/mqtt_io/mqtt_io/server.py", line 127, in _init_module
    install_missing_module_requirements(module)
  File "/home/mqtt_io/mqtt_io/modules/__init__.py", line 53, in install_missing_module_requirements
    raise CannotInstallModuleRequirements(
mqtt_io.exceptions.CannotInstallModuleRequirements: Unable to install packages for module <module 'mqtt_io.modules.gpio.raspberrypi' from '/home/mqtt_io/mqtt_io/modules/gpio/raspberrypi.py'> (['RPi.GPIO']): Command '['/home/mqtt_io/venv/bin/python', '-m', 'pip', 'install', 'RPi.GPIO']' returned non-zero exit status 1.
```

TBobsin, 2024-03-14:
Hello,
i have the same Problem on Raspberry 4 with Debian 12 and Docker setup as described. 
The previous Version was fine. 

The same problem is on the Homeassistant Plugin which belongs to this project. 
Look at this Issue [RuntimeError: This module can only be run on a Raspberry Pi!](https://github.com/hassio-addons/addon-mqtt-io/issues/100)

RezzZ, 2024-05-30:
same problem, no luck getting this to work. Please fix

BenjiU, 2024-05-31:
Hey guys,

I installed gcc in the slim docker in PR #368
Do you have the possiblity to test it from the branch '355-new-version-not-working'?

When I try it with compose file from @Bluscream, it builds rpi.gpio and runs mqtt_io with it on my raspberry:
```
pi@rpi-docker:~/mqtt-io $ docker compose up
WARN[0001] /home/pi/mqtt-io/compose.yaml: `version` is obsolete 
[+] Running 2/2
 ✔ Container mqtt-io-mqtt-io-1                                        Recreated                                                                                                                      11.3s 
 ! mqtt-io Published ports are discarded when using host network mode                                                                                                                                 0.0s 
Attaching to mqtt-io-1
mqtt-io-1  | Collecting RPi.GPIO
mqtt-io-1  |   Downloading RPi.GPIO-0.7.1.tar.gz (29 kB)
mqtt-io-1  |   Preparing metadata (setup.py): started
mqtt-io-1  |   Preparing metadata (setup.py): finished with status 'done'
mqtt-io-1  | Building wheels for collected packages: RPi.GPIO
mqtt-io-1  |   Building wheel for RPi.GPIO (setup.py): started
mqtt-io-1  |   Building wheel for RPi.GPIO (setup.py): finished with status 'done'
mqtt-io-1  |   Created wheel for RPi.GPIO: filename=RPi.GPIO-0.7.1-cp38-cp38-linux_aarch64.whl size=75928 sha256=86912add406817bb4890ae184e0342d5a34b654788316e16a4d4b24a004d264b
mqtt-io-1  |   Stored in directory: /root/.cache/pip/wheels/de/50/02/3fae2f7be5d927c1c4c45d0c94af6b82480cf993194c534321
mqtt-io-1  | Successfully built RPi.GPIO
mqtt-io-1  | Installing collected packages: RPi.GPIO
mqtt-io-1  | Successfully installed RPi.GPIO-0.7.1
mqtt-io-1  | 2024-05-31 22:41:39 mqtt_io.server [INFO] Digital output 'test' current value is False (raw: False)
mqtt-io-1  | 2024-05-31 22:41:39 mqtt_io.server [INFO] Digital input 'button' value changed to False
mqtt-io-1  | 2024-05-31 22:41:39 mqtt_io.server [INFO] Connecting to MQTT...
mqtt-io-1  | 2024-05-31 22:41:40 mqtt_io.server [INFO] Connected to MQTT
mqtt-io-1  | 2024-05-31 22:41:40 mqtt_io.server [INFO] Subscribed to topic: 'pimqttgpio/mydevice/output/test/set'
mqtt-io-1  | 2024-05-31 22:41:40 mqtt_io.server [INFO] Subscribed to topic: 'pimqttgpio/mydevice/output/test/set_on_ms'
mqtt-io-1  | 2024-05-31 22:41:40 mqtt_io.server [INFO] Subscribed to topic: 'pimqttgpio/mydevice/output/test/set_off_ms'
```

**Please test it, so i can merge and release it!**
build docker from the branch with:
`docker buildx build -t flyte/mqtt-io:test .`

@Bluscream stripped compose.yaml:
```
version: '3'
services:
  mqtt-io:
    privileged: true
    user: root
    image: flyte/mqtt-io:test
    ports:
     - "443:443"
     - "1883:1883"
     - "8883:8883"
     - "14567:14567"
    volumes:
      - type: bind
        source: /home/pi/mqtt-io/config.yml
        target: /config.yml
        read_only: true
      - /dev/i2c-0:/dev/i2c-0
      - /dev/mem:/dev/mem
      - /dev/gpiomem:/dev/gpiomem
    devices:
      - /dev/i2c-0
      - /dev/mem
      - /dev/gpiomem
    network_mode: host
```
I ran it with: `docker compose up`

moebius911, 2024-06-02:
hello, same probleme :'(

docker run -ti --rm -v  /home/pi/Docker_Container_data/mqtt-io/config.yml:/config.yml flyte/mqtt-gpio:test
Unable to find image 'flyte/mqtt-gpio:test' locally
docker: Error response from daemon: manifest for flyte/mqtt-gpio:test not found: manifest unknown: manifest unknown.

your image is not found

BenjiU, 2024-06-03:
Hi @moebius911, you have to build it locally first, the mqtt-gpio:test is NOT pushed somewhere.
`docker buildx build -t flyte/mqtt-io:test .` in the branch `355-new-version-not-working`

RezzZ, 2024-06-03:
> 355-new-version-not-working

I get RuntimeError: No access to /dev/mem.  Try running as root!

With my current working mqttio version I don't have to specify any volumes or devices and the container has access to /dem/mem

closed 2024-06-19 by BenjiU

Bluscream, 2024-06-19:
@BenjiU when is #386 going to be on docker?

BenjiU, 2024-06-19:
I'm  working on the release ;-)