flyte/mqtt-io #356: Wont Connect to MQTT Broker and getting error in log from python script

Arsenal10108, 2024-03-08:
- [ X] I have read the [README.md](https://github.com/flyte/pi-mqtt-gpio/blob/develop/README.md) file and visited the [Documentation](https://mqtt-io.app/) to see if the information there helps.

**Describe the bug**
Wont Connect to MQTT Broker and getting error in logs

**Expected behavior**
for it to work :)

**Error messages and traceback**
```
Exception ignored in: <function Client.__del__ at 0x75b7b610>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 874, i                                                                             n __del__
    self._reset_sockets()
  File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 1133,                                                                              in _reset_sockets
    self._sock_close()
  File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 1119,                                                                              in _sock_close
    if not self._sock:
AttributeError: 'Client' object has no attribute '_sock'
2024-03-07 22:46:04,093 mqtt_io.server (ERROR): Exception in critical task:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/server.py", line 1380, in                                                                              _main_loop
    await self._connect_mqtt()
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/server.py", line 702, in                                                                              _connect_mqtt
    self.mqtt = AbstractMQTTClient.get_implementation(config["client_module"])(
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/mqtt/asyncio_mqtt.py", li                                                                             ne 91, in __init__
    self._client = Client(
  File "/usr/local/lib/python3.9/dist-packages/asyncio_mqtt/client.py", line 34,                                                                              in __init__
    self._client = mqtt.Client(client_id=client_id, protocol=protocol, clean_ses                                                                             sion=clean_session, transport=transport)
TypeError: __init__() missing 1 required positional argument: 'callback_api_vers                                                                             ion'
2024-03-07 22:46:06,098 mqtt_io.server (INFO): Reconnecting to MQTT broker (infi                                                                             nite retries remaining)
```

**Config**
```
mqtt:
  host: 192.168.1.246
  port: 1883
  user: "mosquitto"
  password: ""
  topic_prefix: home

gpio_modules:
  - name: raspberrypi
    module: raspberrypi
    cleanup: no  # This optional boolean value sets whether the module's `cleanup()` function will be called when the software exits.

digital_inputs:
#Big Garage Door
  - name: biggaragedoor
    module: raspberrypi
    pin: 5
    on_payload: "ON"
    off_payload: "OFF"
    pullup: yes
    pulldown: no
#Little Garage Door
  - name: littlegaragedoor
    module: raspberrypi
    pin: 17
    on_payload: "ON"
    off_payload: "OFF"
    pullup: yes
    pulldown: no
#Front Garage  Window
  - name: garagefrontwindow
    module: raspberrypi
    pin: 18
    on_payload: "ON"
    off_payload: "OFF"
    pullup: yes
    pulldown: no
#Rear Garage Window
  - name: garagerearwindow
    module: raspberrypi
    pin: 27
    on_payload: "ON"
    off_payload: "OFF"
    pullup: yes
    pulldown: no
    
#OUTPUTS##############
#Garage Heat
digital_outputs:
  - name: garageheat
    module: raspberrypi
    pin: 21
    on_payload: "ON"
    off_payload: "OFF"
    initial: low
#Garage Door Opener
  - name: garagedooropener
    module: raspberrypi
    pin: 24
    on_payload: "ON"
    off_payload: "OFF"
    initial: low

######SENSORS############

sensor_modules:

  - name: bh1750
    module: bh1750
    i2c_bus_num: 1
    chip_addr: 0x23

  - name: lm75
    module: lm75
    i2c_bus_num: 1
    chip_addr: 0x48

sensor_inputs:
  - name: temperature
    module: lm75
    interval: 15
    digits: 4

  - name: bh1750_lux
    module: bh1750
    interval: 10
    digits: 2

logging:
  version: 1
  formatters:
    simple:
      format: "%(asctime)s %(name)s (%(levelname)s): %(message)s"
  handlers:
    console:
      class: logging.StreamHandler
      level: INFO
      formatter: simple
      stream: ext://sys.stdout
  loggers:
    mqtt_io:
      level: INFO
      handlers: [console]
      propagate: yes

```

**Hardware**
 - Platform: [Raspberry Pi, ]
 - Connected hardware:  Nope

**System:**
 - OS: [Raspbian]
 - Python version: [python 3.9.2]
 - User you're running as: [pi]
 - Using a virtualenv?: yes

**Additional context**
Add any other context about the problem here.

BenjiU, 2024-03-08:
Duplikate of #348 
Please see work around there.

Arsenal10108, 2024-03-08:
Whoops Didn't see that! Thanks BenjiU

closed 2024-03-08 by Arsenal10108