flyte/mqtt-io #357: DHT22 Adafruit_DHT problem

alexfrench, 2024-03-13:
I am setting up mqtt-io on a new RPi. Previously this has been very simple and works well. Thank you! I'm hoping someone might be able to offer a solution for my current issue.

Previously, I followed instructions and downloaded the repo, then from the repo folder I ran: `python -m pip install .` The latest version suggests using `pip3 install mqtt-io ` instead. However, this gives an error:

> error: externally-managed-environment
> 
> × This environment is externally managed
> ╰─> To install Python packages system-wide, try apt install
>     python3-xyz, where xyz is the package you are trying to
>     install.
> 
>     If you wish to install a non-Debian-packaged Python package,
>     create a virtual environment using python3 -m venv path/to/venv.
>     Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
>     sure you have python3-full installed.
> 
>     For more information visit http://rptl.io/venv
> 
> note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
> hint: See PEP 668 for the detailed specification.
> 

I settled on using the below which seems to work around the install issues:
```
pip3 install mqtt-io==2.2.9 --break-system-packages
pip3 install paho-mqtt==1.6.1 --break-system-packages

```

The install seems to work until I add a DHT22 sensor (working fine on earlier installs) when I see the following:

2024-03-13 03:24:17 mqtt_io.__main__ [ERROR   ] MqttIo crashed!
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.11/site-packages/mqtt_io/modules/__init__.py", line 51, in install_missing_module_requirements
    install_missing_requirements(pkgs_required)
  File "/home/pi/.local/lib/python3.11/site-packages/mqtt_io/modules/__init__.py", line 22, in install_missing_requirements
    check_call([sys.executable, "-m", "pip", "install"] + pkgs_required)
  File "/usr/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python', '-m', 'pip', 'install', 'Adafruit_DHT']' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.11/site-packages/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/home/pi/.local/lib/python3.11/site-packages/mqtt_io/server.py", line 1450, in run
    self._init_sensor_modules()
  File "/home/pi/.local/lib/python3.11/site-packages/mqtt_io/server.py", line 284, in _init_sensor_modules
    self.sensor_modules[sens_config["name"]] = _init_module(
                                               ^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/mqtt_io/server.py", line 127, in _init_module
    install_missing_module_requirements(module)
  File "/home/pi/.local/lib/python3.11/site-packages/mqtt_io/modules/__init__.py", line 53, in install_missing_module_requirements
raise CannotInstallModuleRequirements(
mqtt_io.exceptions.CannotInstallModuleRequirements: Unable to install packages for module <module 'mqtt_io.modules.sensor.dht22' from '/home/pi/.local/lib/python3.11/site-packages/mqtt_io/modules/sensor/dht22.py'> (['Adafruit_DHT']): Command '['/usr/bin/python', '-m', 'pip', 'install', 'Adafruit_DHT']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.11/site-packages/mqtt_io/modules/__init__.py", line 51, in install_missing_module_requirements
    install_missing_requirements(pkgs_required)
  File "/home/pi/.local/lib/python3.11/site-packages/mqtt_io/modules/__init__.py", line 22, in install_missing_requirements
    check_call([sys.executable, "-m", "pip", "install"] + pkgs_required)
  File "/usr/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python', '-m', 'pip', 'install', 'Adafruit_DHT']' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/pi/.local/lib/python3.11/site-packages/mqtt_io/__main__.py", line 115, in <module>
    main()
File "/home/pi/.local/lib/python3.11/site-packages/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/home/pi/.local/lib/python3.11/site-packages/mqtt_io/server.py", line 1450, in run
    self._init_sensor_modules()
  File "/home/pi/.local/lib/python3.11/site-packages/mqtt_io/server.py", line 284, in _init_sensor_modules
    self.sensor_modules[sens_config["name"]] = _init_module(
                                               ^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/mqtt_io/server.py", line 127, in _init_module
    install_missing_module_requirements(module)
  File "/home/pi/.local/lib/python3.11/site-packages/mqtt_io/modules/__init__.py", line 53, in install_missing_module_requirements
    raise CannotInstallModuleRequirements(
mqtt_io.exceptions.CannotInstallModuleRequirements: Unable to install packages for module <module 'mqtt_io.modules.sensor.dht22' from '/home/pi/.local/lib/python3.11/site-packages/mqtt_io/modules/sensor/dht22.py'> (['Adafruit_DHT']): Command '['/usr/bin/python', '-m', 'pip', 'install', 'Adafruit_DHT']' returned non-zero exit status 1.

BenjiU, 2024-03-13:
I guess you get the same error as above, when mqtt tries to install packages...

are you in a virtual environment? did you forget to source it? 
`source /home/BENUTZER/venv-test/bin/activate`

Or is something broken on your PI?

The error tells you, that you install a package systemwide in an environment (venv?) and it warns you, if you really want to do this.
And you exactly do this:
`You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.`

After that, mqtt is also not able to install packages.

So i guess, the error is somewhere in the python setup and not mqtt related.

What distro/version have you installed on your new rpi? May you try with an older release?

Merlinux-source, 2024-05-07:
same issue here on a raspberry pi using docker.
I'm trying docker because I experienced the same issue as the issue creator using a venv. because I wanted to resolve the it doesnt work on my machine issue, I ran it in docker.
maybe my compose file is wrong, but I don't think so [the converter](https://www.composerize.com/) works verry reliably
I'm using raspberrypi os lite on a Raspberry Pi 4B rev 1.2
the content of /proc/version is
`Linux version 6.1.0-rpi8-rpi-v8 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25)`
I didn't install any pip package outside of a virtual environment to prevent this from happening I installed every service using docker. this pi is updated and upgraded daily and runs the stable branch. 

I'm using a docker compose file:
```yml
name: mqttio
services:
    mqtt-gpio:
        tty: true
        stdin_open: true
        volumes:
            - /etc/mqttio/mqttio-config.yaml:/config.yml
        privileged: true
        image: flyte/mqtt-gpio:armv7l
```
my config.yml looks like this:
```yml
mqtt:
  host: 192.168.178.95 
  topic_prefix: pflanzen/sensoren # change this as you see fit, I like [room]/[device]
  status_payload_running: available
  status_payload_stopped: unavailable
  status_payload_dead: unavailable

sensor_modules:
- name: dht22_sensor
  module: dht22
  type: dht22
  pin: 2 # this may change if you add additional sensors or don't use pin 4 from above

sensor_inputs:
- name: pflanzen_temperature
  module: dht22_sensor
  digits: 2
  interval: 5
  type: temperature
- name: pflanzen_humidity
  module: dht22_sensor
  digits: 2
  interval: 5
  type: humidity
```

```
root@pflanzen:/opt/mqttio# docker compose down
[+] Running 2/2
 ✔ Container mqttio-mqtt-gpio-1  Removed                                                                                                                                                                                                0.0s 
 ✔ Network mqttio_default        Removed                                                                                                                                                                                                0.3s 
root@pflanzen:/opt/mqttio# docker compose up
[+] Running 2/2
 ✔ Network mqttio_default        Created                                                                                                                                                                                                0.1s 
 ✔ Container mqttio-mqtt-gpio-1  Created                                                                                                                                                                                                0.2s 
Attaching to mqtt-gpio-1
mqtt-gpio-1  | 2024-05-07 00:03:03,905 mqtt_gpio (INFO): Startup
mqtt-gpio-1  | ERROR: Exception:
mqtt-gpio-1  | Traceback (most recent call last):
mqtt-gpio-1  |   File "/home/mqttgpio/.local/share/virtualenvs/mqttgpio-815CbdL_/lib/python3.6/site-packages/pip/_internal/cli/base_command.py", line 186, in _main
mqtt-gpio-1  |     status = self.run(options, args)
mqtt-gpio-1  |   File "/home/mqttgpio/.local/share/virtualenvs/mqttgpio-815CbdL_/lib/python3.6/site-packages/pip/_internal/commands/install.py", line 289, in run
mqtt-gpio-1  |     wheel_cache = WheelCache(options.cache_dir, options.format_control)
mqtt-gpio-1  |   File "/home/mqttgpio/.local/share/virtualenvs/mqttgpio-815CbdL_/lib/python3.6/site-packages/pip/_internal/cache.py", line 291, in __init__
mqtt-gpio-1  |     self._ephem_cache = EphemWheelCache(format_control)
mqtt-gpio-1  |   File "/home/mqttgpio/.local/share/virtualenvs/mqttgpio-815CbdL_/lib/python3.6/site-packages/pip/_internal/cache.py", line 267, in __init__
mqtt-gpio-1  |     self._temp_dir = TempDirectory(kind="ephem-wheel-cache")
mqtt-gpio-1  |   File "/home/mqttgpio/.local/share/virtualenvs/mqttgpio-815CbdL_/lib/python3.6/site-packages/pip/_internal/utils/temp_dir.py", line 116, in __init__
[... 5 lines cut ...]
mqtt-gpio-1  | OSError: [Errno 28] No space left on device: '/tmp/pip-ephem-wheel-cache-8qar9x8q'
mqtt-gpio-1  | Traceback (most recent call last):
mqtt-gpio-1  |   File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 264, in install_missing_requirements
mqtt-gpio-1  |     check_call(["/usr/bin/env", "pip", "install"] + pkgs_required)
mqtt-gpio-1  |   File "/usr/local/lib/python3.6/subprocess.py", line 311, in check_call
mqtt-gpio-1  |     raise CalledProcessError(retcode, cmd)
mqtt-gpio-1  | subprocess.CalledProcessError: Command '['/usr/bin/env', 'pip', 'install', 'Adafruit_DHT']' returned non-zero exit status 2.
mqtt-gpio-1  | 
mqtt-gpio-1  | During handling of the above exception, another exception occurred:
mqtt-gpio-1  | 
mqtt-gpio-1  | Traceback (most recent call last):
mqtt-gpio-1  |   File "/usr/local/lib/python3.6/runpy.py", line 193, in _run_module_as_main
mqtt-gpio-1  |     "__main__", mod_spec)
mqtt-gpio-1  |   File "/usr/local/lib/python3.6/runpy.py", line 85, in _run_code
mqtt-gpio-1  |     exec(code, run_globals)
mqtt-gpio-1  |   File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 901, in <module>
mqtt-gpio-1  |     main(args)
mqtt-gpio-1  |   File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 787, in main
mqtt-gpio-1  |     SENSOR_MODULES[sensor_config["name"]] = configure_sensor_module(sensor_config)
[... cut ...]
mqtt-gpio-1  |     install_missing_requirements(sensor_module)
mqtt-gpio-1  |   File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 268, in install_missing_requirements
mqtt-gpio-1  |     % (module, pkgs_required, err)
mqtt-gpio-1  | __main__.CannotInstallModuleRequirements: Unable to install packages for module <module 'pi_mqtt_gpio.modules.dht22' from '/home/mqttgpio/pi_mqtt_gpio/modules/dht22.py'> (['Adafruit_DHT']): Command '['/usr/bin/env', 'pip', 'install', 'Adafruit_DHT']' returned non-zero exit status 2.
mqtt-gpio-1 exited with code 1```

BenjiU, 2024-06-03:
`OSError: [Errno 28] No space left on device: '/tmp/pip-ephem-wheel-cache-8qar9x8q'`
I'll try to reproduce on my raspberry 3B (currently i do not have a pi 4).