flyte/mqtt-io #359: Crash on startup

openstep, 2024-03-22:
- [ ] I have read the [README.md](https://github.com/flyte/pi-mqtt-gpio/blob/develop/README.md) file and visited the [Documentation](https://mqtt-io.app/) to see if the information there helps.

**Describe the bug**
When starting with the config below, there is an error that prevents the mqtt-io from working

**Expected behavior**
A working code

**Error messages and traceback**
ERROR:
`python3 -m mqtt_io config.yml
2024-03-22 19:07:15 mqtt_io.server [INFO] Digital output 'rele1' current value is False (raw: False)
2024-03-22 19:07:15 mqtt_io.server [INFO] Digital output 'rele2' current value is False (raw: False)
2024-03-22 19:07:16 mqtt_io.server [INFO] Digital output 'rele3' current value is False (raw: False)
2024-03-22 19:07:17 mqtt_io.server [INFO] Digital output 'rele4' current value is False (raw: False)
2024-03-22 19:07:17 mqtt_io.server [INFO] Digital output 'rele5' current value is False (raw: False)
2024-03-22 19:07:18 mqtt_io.server [INFO] Digital output 'rele6' current value is False (raw: False)
2024-03-22 19:07:19 mqtt_io.server [INFO] Digital output 'rele7' current value is False (raw: False)
2024-03-22 19:07:20 mqtt_io.server [INFO] Digital output 'rele8' current value is False (raw: False)
Exception ignored in: <function Client.__del__ at 0x75d3b198>
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/home/pi/.local/lib/python3.7/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/home/pi/.local/lib/python3.7/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
AttributeError: 'Client' object has no attribute '_sock'
2024-03-22 19:07:20 mqtt_io.server [ERROR] Exception in critical task:
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 1380, in _main_loop
    await self._connect_mqtt()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 703, in _connect_mqtt
    self.mqtt_client_options
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/mqtt/asyncio_mqtt.py", line 101, in __init__
    clean_session=options.clean_session,
  File "/home/pi/.local/lib/python3.7/site-packages/asyncio_mqtt/client.py", line 34, in __init__
    self._client = mqtt.Client(client_id=client_id, protocol=protocol, clean_session=clean_session, transport=transport)
TypeError: __init__() missing 1 required positional argument: 'callback_api_version'
2024-03-22 19:07:22 mqtt_io.server [INFO] Reconnecting to MQTT broker (infinite retries remaining)`

**Config**
```
mqtt:
  host: 192.168.1.40
  port: 1883
  topic_prefix: home
  ha_discovery:
    enabled: yes
    name: waterberry

# GPIO
gpio_modules:
  # Use the Raspberry Pi built-in GPIO
  - name: rpi
    module: raspberrypi

digital_outputs:
  - name: rele1
    module: rpi
    pin: 18

  - name: rele2
    module: rpi
    pin: 23
```

**Hardware**
Raspberry pi 3b+  and the mqtt server is on another host

**System:**
 - Raspbian GNU/Linux 10, 
 - Python version:  Python 3.7.3 
 - User you're running as: pi
 - Using a virtualenv?: no

chrislp, 2024-03-25:
Hi have the same issue, same system and the following configuration:
```yaml
mqtt:
  host: 192.168.xx
  port: 1883
  topic_prefix: mqtt_io
  ha_discovery:
    enabled: yes

sensor_modules:
  - name: dht22_sensor
    module: dht22
    type: DHT22
    pin: 2

sensor_inputs:
  - name: temperature
    module: dht22_sensor
    interval: 10
    digits: 4
    type: temperature

  - name: humidity
    module: dht22_sensor
    interval: 10
    digits: 4
    type: humidity
```

alext, 2024-03-29:
This looks like #348 - downgrading `paho-mqtt` to the latest 1.x version should solve it.

CoherentLogic, 2024-04-09:
how do you do that?

alext, 2024-04-13:
> how do you do that?

If you installed it using pip, then you can downgrade with `pip install 'paho-mqtt<2'`, which will get you the latest 1.x version.

chrislp, 2024-04-15:
Downgrading `paho-mqtt` did the trick for me. But I had to upgrade my ancient OS to Raspbian 11 first.

BenjiU, 2024-06-02:
will be fixed with PR: #364

closed 2024-07-30 by BenjiU