flyte/mqtt-io #373: Adafruit_DHT (DHT22, ...) package is deprecated  on Raspberry Pi

eMindFfM, 2024-06-19:
Adafruit_DHT (DHT22, ...) package is deprecated  on Raspberry Pi  and can not be installed any longer, thus the package can not run.

an alternative is package provided  adafruit-circuitpython-dht

Source: https://github.com/adafruit/Adafruit_CircuitPython_DHT

Can you please update

mschlenstedt, 2024-06-20:
Installed via apt or pip? Installation should still be possible via pip.

eMindFfM, 2024-06-20:
The connection to the mqtt server seem to work
[image]

eMindFfM, 2024-06-21:
Update ... The issue with the Websocket Wrapper can be soved by installing an old version of paho-mqtt

`pip install paho-mqtt==1.6.1`

eMindFfM, 2024-06-21:
The Issue with `RuntimeError: Unknown platform. `  remains