flyte/mqtt-io #399: config not found - new install via addons in home assistant

ptryk, 2024-09-03:
- [ ] I have read the [README.md](https://github.com/flyte/pi-mqtt-gpio/blob/develop/README.md) file and visited the [Documentation](https://mqtt-io.app/) to see if the information there helps.

**Describe the bug**
system  does not find the config file, created and tried in a number of ways - my best guess is that the root /homeautomation/ is confusing it but I have tried creating the following directories and files and get the same error regardless: 
/config/mqtt-io/config.yml
config/mqtt-io/config.yml
/homeassistant/config/mqtt-io/config.yml
homeassistant/config/mqtt-io/config.yml
/homeassistant/config/mqtt-io_config.yml
etc.. 

only able to get it to work by putty'ng in and moving the config to the base directory /config/.. .yml
then it seems to read it.  .. a bit unfortunate that can not use the base tools 

**Expected behavior**
the system should find the config file set within home assistant and then hopefully start .. sadly so far it does not

**Error messages and traceback**
```
-----------------------------------------------------------
 Add-on: MQTT IO
 Expose GPIO modules and digital sensors via MQTT for remote control and monitoring.
-----------------------------------------------------------
 Add-on version: 0.4.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.4  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.8.1
 Home Assistant Supervisor: 2024.08.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
Log level is set to TRACE
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service mqtt-io: starting
s6-rc: info: service mqtt-io successfully started
s6-rc: info: service legacy-services: starting
[07:38:49] TRACE: bashio::config: configuration_file
[07:38:49] TRACE: bashio::addon.config
[07:38:49] TRACE: bashio::cache.exists: addons.self.options.config
[07:38:49] TRACE: bashio::fs.file_exists: /tmp/.bashio/addons.self.options.config.cache
s6-rc: info: service legacy-services successfully started
[07:38:49] TRACE: bashio::cache.get: addons.self.options.config
[07:38:49] TRACE: bashio::cache.exists: addons.self.options.config
[07:38:49] TRACE: bashio::fs.file_exists: /tmp/.bashio/addons.self.options.config.cache
[07:38:49] TRACE: bashio::jq: {"configuration_file":"/homeassistant/config/mqtt-io_config.yml","log_level":"trace"} if (.configuration_file == null) then
            null
        elif (.configuration_file | type == "string") then
            .configuration_file // empty
        elif (.configuration_file | type == "boolean") then
            .configuration_file // false
        elif (.configuration_file | type == "array") then
            if (.configuration_file == []) then
                empty
            else
                .configuration_file[]
            end
        elif (.configuration_file | type == "object") then
            if (.configuration_file == {}) then
                empty
            else
                .configuration_file
            end
        else
            .configuration_file
        end
[07:38:49] TRACE: bashio::fs.file_exists: /homeassistant/config/mqtt-io_config.yml
[07:38:49] FATAL: 
[07:38:49] FATAL: Seems like the configured configuration file does
[07:38:49] FATAL: not exists:
[07:38:49] FATAL: 
[07:38:49] FATAL: /homeassistant/config/mqtt-io_config.yml
[07:38:49] FATAL: 
[07:38:49] FATAL: Please check the add-on configuration, or create the
[07:38:49] FATAL: configuration file. More information about the
[07:38:49] FATAL: configuration file format see:
[07:38:49] FATAL: 
[07:38:49] FATAL: https://mqtt-io.app
[07:38:49] FATAL: 
[07:38:49] TRACE: bashio::exit.nok:
[07:38:49] TRACE: bashio::var.has_value: 
[07:38:49] INFO: Service MQTT IO exited with code 1 (by signal 0)
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service mqtt-io: stopping
s6-rc: info: service mqtt-io successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

**Config**
```yaml
Your YAML config file. Please remove any passwords and sensitive hostnames/IP addresses.
```

**Hardware**
 - Platform: Raspberry Pi, 
 - Connected hardware:  trying to get the home assistant to read from a GPIO pin in this case a hall effect pulse 

**System:**
 - HA .. mqtt-io says version within the home assistant as : MQTT IO Current version: 0.4.0
 - Python version:  Python 3.12.3
 - User you're running as: home assistant "owner"
 - Using a virtualenv?: no (i do not think ha is considered a virtual env)

**Additional context**
loaded directly from the add in on the home assistant.

BenjiU, 2024-09-17:
Hi @ptryk,

I have zero experience with homeassistant, but I guess that this is not a problem of mqtt-io, but related to the homeassistant plugin, right?
mqtt-io just gets the file name on startup and does not know anything about homeassistant environment.

Anyone who can confirm or correct this?

BenjiU, 2024-09-25:
closed, not related to mqtt-io module

closed 2024-09-25 by BenjiU