flyte/mqtt-io #400: Add support for `XL9535` relay board

mahnunchik, 2024-09-16:
**Is your feature request related to a problem? Please describe.**

It would be nice if mqtt-io supports `XL9535` relay board.

**Describe the solution you'd like**

Normal integration like other I2C `PCF8574` or `PCF8575` gpio expanders.

**Describe alternatives you've considered**

Write the code...

**Additional context**

[image]

I bought a great relay board, with 16 relays, which is enough for all my needs. I would like to have **simple** and **convenient** integration.

More info:
* Python module and description https://github.com/mcauser/micropython-xl9535-kxv5-relay
* Datasheet https://www.lcsc.com/datasheet/lcsc_datasheet_2211110930_XINLUDA-XL9535_C561273.pdf
* ESPHome integration https://github.com/esphome/esphome/tree/857a3dcf72d74fa53fba2f6c63f2ceaaec3e5e59/esphome/components/xl9535
* Discussion on micropython https://github.com/orgs/micropython/discussions/13505

mahnunchik, 2024-09-18:
It seems `XL9535` is logically the same chip as `PCA9535/TCA9535`

mahnunchik, 2024-09-20:
Hi @flyte @mschlenstedt @BenjiU

I'd like to work on support of `XL9535/PCA9535/TCA9535` chip and prepare PR. Could you please point me how to add a new device?

It seems there is no existing python "driver" published in pip. Could low level code be placed in this repo or should I make separate package?

---

Some kind of nodejs driver https://github.com/isky04/node-red-contrib-xl9535

mschlenstedt, 2024-09-21:
I recently started to add some sensors to MQTTIO. As far as I know there's no developer documentation. I looked at the existing modules, take one which is similar to the one I would like to add and started and tested on a life system.

You'll finde the code here:

Sensors: https://github.com/flyte/mqtt-io/tree/develop/mqtt_io/modules/sensor
Input/Outputs: https://github.com/flyte/mqtt-io/tree/develop/mqtt_io/modules/gpio