flyte/mqtt-io #416: Does not run on Raspberry Pi 4 aarch64 architecture using Docker (ModuleNotFoundError: No module named 'RPi' error)

megapearl, 2024-10-30:
- [V] I have read the [README.md](https://github.com/flyte/pi-mqtt-gpio/blob/develop/README.md) file and visited the [Documentation](https://mqtt-io.app/) to see if the information there helps.

**Describe the bug**
When running by docker at Aarch64 architecture on a raspberry pi 4 it gives an error upon startup.

**Expected behavior**
Should run on aarch64 architecture

**Error messages and traceback**
```
ModuleNotFoundError: No module named 'RPi' error
```

**Config**
```yaml
N/A it does not even start.
```

**Hardware**
 - Platform: [Raspberry Pi4 ]
 - Connected hardware:  [N/A]

**System:**
 - OS: [Linux cupboard-dockers 6.6.56-0-rpi #1-Alpine SMP PREEMPT Fri Oct 11 07:01:26 UTC 2024 aarch64 Linux]
 - Python version: [Python 3.11.10]
 - User you're running as: [root]
 - Using a virtualenv?: [No, running in docker]

**Additional context**
When I add 'RPi-GPIO' to pyproject.toml AND 'pip install RPi.GPIO adafruit-blinka' to Dockerfile using:

        sed -i '/^\[tool.poetry.dependencies\]/a RPi-GPIO = "^0.7.1"' pyproject.toml
        sed -i '/COPY pyproject.toml .\//a RUN pip install RPi.GPIO adafruit-blinka' Dockerfile
 
And rebuild the docker it runs fine on my aarch64 architecture raspberry pi4

tkislan, 2025-02-02:
I have this same problem

```
gpio_modules:
  - name: raspberrypi
    module: raspberrypi

sensor_modules:
  - name: dht22_sensor
    module: dht22
    type: AM2302
    pin: 4

  - name: mhz19
    module: mhz19
    device: "/dev/ttyAMA0"
    range: 5000

sensor_inputs:
  - name: temperature
    module: dht22_sensor
    interval: 300
    digits: 4
    type: temperature

  - name: humidity
    module: dht22_sensor
    interval: 300
    digits: 4
    type: humidity

  - name: co2_mhz19
    module: mhz19
    interval: 300
    ha_discovery:
      device_class: carbon_dioxide
```

when I specify gpio_module for raspberry, it complaints not having any digital inputs or outputs
```
Config did not validate due to errors in the following sections:                                                                │
│ gpio_modules:                                                                                                                   │
│   raspberrypi:                                                                                                                  │
│   - GPIO module 'raspberrypi' does not have any digital_input or digital_output configs
```

and if I remove it, I get an error that it can't import RPi GPIO

```
  File "/home/mqtt_io/venv/lib/python3.8/site-packages/adafruit_blinka/microcontroller/bcm283x/pin.py", line 5, in <module>
    from RPi import GPIO
ModuleNotFoundError: No module named 'RPi'
```

I don't think that the first case should be an error that shuts down the whole process, and it doesn't run at all .. a warning should be sufficient, as I don't think there would be any problem running with some extra config

or add extra configuration option, to allow to add extra dependencies to be installed

I would like to avoid building a custom Docker image just for this

tkislan, 2025-02-02:
Hmm, but there might be another problem, because even with manually adding downloading the package, it doesn't report any values
```
/bin/sh -ec "venv/bin/python -m pip install rpi.gpio && venv/bin/python -m mqtt_io /config.yml"
```