flyte/mqtt-io #420: module `ds18b` dynamic output without address

petr-stupka, 2025-01-28:
My plan is to create configuration template (universal) so hardcoding value in config or scripting around may be not so ideal. 

Is possible the to remove `address` from `sensor_modules` and the `sensor_inputs` will produce output as array (all sensors values) and we may use HA (or any other automation) to filter it out?

Please check this script which is doing exactly that: https://github.com/Pekaway/VAN_PI/blob/main/VanPi-Core-OS/data/ds18b20_py/ds18b20.py

```yaml
sensor_modules:
  - name: ds18b20
    module: ds18b
    type: DS18B20

sensor_inputs:
  - name: temperatures
    module: ds18b20
    interval: 5
    digits: 2
```

Alternative can be to move `address` from `sensor_modules` to `sensor_inputs`? Then it will be easier to autogenerate the `sensor_inputs`

So instead of this

```yaml
sensor_modules:
  - name: ds18b20
    module: ds18b
    type: DS18B20
    address: "1a19709614ff"

sensor_inputs:
  - name: temp1
    module: ds18b20
    interval: 5
    digits: 2
```

To have this

```yaml
sensor_modules:
  - name: ds18b20
    module: ds18b
    type: DS18B20

sensor_inputs:
  - name: temp1
    module: ds18b20
    interval: 5
    digits: 2
    address: "1a19709614ff"
```

Thank you

BenjiU, 2025-02-03:
Hi @petr-stupka,

okay, this is a very special issue.
It's a matter of taste:
- To your first suggestion: this app was designed to do the most calculation in the app (on your edge node) and not in your (central) automation software (like HA).
- The second suggestion would reduce your code a little bit, but is currently not changeable due to the software structure. A sensor_module is a piece of hardware and the sensor_inputs is/are the value(s) you can get of that hardware. It's some kind of decoupling. On the other hand it could cause confusion for sensors that have two values (e.g. temp and humidity): both sensor_inputs must then have the same address.

Feel free to integrate your changes via a PR but please take care that you do not break the current config files of other users :-D

If you don't mind, I would like to close this issue, as it is not doable currently. But your PR is welcome of course.

Greetings!
Benni

petr-stupka, 2025-02-03:
Thank you @BenjiU. You can close the issue as it answered the question, no problem, and i will come back with PR

closed 2025-02-03 by BenjiU