flyte/mqtt-io #424: Module mhz19 not found (python3.7, RP4 64bit)

mareklab, 2025-02-23:
Hi, I hope you can help... 🙏 

Trying to use mqtt-io for my MH-Z19 sensor since HomeAssistant no longer supports it directly, and mqtt-io seems to be a good fit.

So first I've made sure the MH-Z19 module is installed (it wasn't easy, also 'mhz19' was not available):
```
$ pip3 install mh-z19
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: mh-z19 in ./.local/lib/python3.7/site-packages (3.1.6)
Requirement already satisfied: getrpimodel>=0.1.26 in ./.local/lib/python3.7/site-packages (from mh-z19) (0.1.26)
Requirement already satisfied: pyserial in /usr/lib/python3/dist-packages (from mh-z19) (3.4)
Requirement already satisfied: requests in ./.local/lib/python3.7/site-packages (from mh-z19) (2.31.0)
Collecting argparse (from mh-z19)
  Using cached https://www.piwheels.org/simple/argparse/argparse-1.4.0-py2.py3-none-any.whl (23 kB)
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from mh-z19) (0.7.0)
Requirement already satisfied: gpiozero in /usr/lib/python3/dist-packages (from mh-z19) (1.6.2)
Requirement already satisfied: lgpio in /usr/local/lib/python3.7/dist-packages/lgpio-0.2.2.0-py3.7-linux-aarch64.egg (from mh-z19) (0.2.2.0)
Requirement already satisfied: charset-normalizer<4,>=2 in ./.local/lib/python3.7/site-packages (from requests->mh-z19) (3.2.0)
Requirement already satisfied: idna<4,>=2.5 in /usr/lib/python3/dist-packages (from requests->mh-z19) (2.6)
Requirement already satisfied: urllib3<3,>=1.21.1 in ./.local/lib/python3.7/site-packages (from requests->mh-z19) (1.26.16)
Requirement already satisfied: certifi>=2017.4.17 in ./.local/lib/python3.7/site-packages (from requests->mh-z19) (2023.7.22)
Installing collected packages: argparse
Successfully installed argparse-1.4.0
```

Then I've made sure the `mqtt-io` is there:
```
$ pip3 install mqtt-io
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: mqtt-io in ./.local/lib/python3.7/site-packages (2.3.0)
Requirement already satisfied: Cerberus<2.0.0,>=1.3.2 in ./.local/lib/python3.7/site-packages (from mqtt-io) (1.3.7)
Requirement already satisfied: PyYAML<7.0.0,>=6.0.1 in ./.local/lib/python3.7/site-packages (from mqtt-io) (6.0.1)
Requirement already satisfied: asyncio-mqtt<0.9.0,>=0.8.1 in ./.local/lib/python3.7/site-packages (from mqtt-io) (0.8.1)
Requirement already satisfied: backoff<2.0.0,>=1.10.0 in /usr/local/lib/python3.7/dist-packages (from mqtt-io) (1.10.0)
Requirement already satisfied: confp<0.5.0,>=0.4.0 in ./.local/lib/python3.7/site-packages (from mqtt-io) (0.4.1)
Requirement already satisfied: typing-extensions<4.0.0,>=3.7.4 in /usr/local/lib/python3.7/dist-packages (from mqtt-io) (3.7.4.3)
Requirement already satisfied: paho-mqtt>=1.5.0 in /usr/local/lib/python3.7/dist-packages (from asyncio-mqtt<0.9.0,>=0.8.1->mqtt-io) (1.6.1)
Requirement already satisfied: importlib-metadata in ./.local/lib/python3.7/site-packages (from Cerberus<2.0.0,>=1.3.2->mqtt-io) (4.13.0)
Requirement already satisfied: jinja2 in /usr/lib/python3/dist-packages (from confp<0.5.0,>=0.4.0->mqtt-io) (2.10)
Requirement already satisfied: zipp>=0.5 in /usr/local/lib/python3.7/dist-packages (from importlib-metadata->Cerberus<2.0.0,>=1.3.2->mqtt-io) (3.4.0)

```

But, when trying to run it, I keep gettting:

```
$ python -m pi_mqtt_gpio.server mqtt-io.yaml
2025-02-23 18:11:36,401 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/.local/lib/python3.7/site-packages/pi_mqtt_gpio/server.py", line 1269, in <module>
    main(args)
  File "/home/pi/.local/lib/python3.7/site-packages/pi_mqtt_gpio/server.py", line 1085, in main
    SENSOR_MODULES[sensor_config["name"]] = configure_sensor_module(sensor_config)
  File "/home/pi/.local/lib/python3.7/site-packages/pi_mqtt_gpio/server.py", line 580, in configure_sensor_module
    sensor_module = import_module("pi_mqtt_gpio.modules.%s" % sensor_config["module"])
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'pi_mqtt_gpio.modules.mh-z19'
```

**Config**
```

mqtt:
  host: 192.168.my.ip
  port: 1888
  user: [-]
  password: [-]
  topic_prefix: mqtt_io

sensor_modules:
  - name: mhz19
    module: mh-z19
    device: "/dev/ttyUSB_CO2"
    range: 5000
```

**Hardware**
 - Platform: Raspberry Pi 4i 
 - Connected hardware: MH-Z19 via USB to RP4
 
**System:**
 - OS: [e.g. Raspbian] `Linux pi-eHouse 6.6.30-v8+ #1763 SMP PREEMPT Mon May 13 17:50:31 BST 2024 aarch64 GNU/Linux`
 - Python version: 3.7.3
 - User you're running as: pi
 - Using a virtualenv?: no

BenjiU, 2025-02-23:
Hi @mareklab,

the module in the config should be mhz19 not mh-z19:
```
sensor_modules:
  - name: mhz19
    module: mh-z19
    device: "/dev/ttyUSB_CO2"
    range: 5000
```

https://github.com/flyte/mqtt-io/blob/develop/mqtt_io/modules/sensor/mhz19.py

mareklab, 2025-02-23:
Thanks, I've tried that before - but the module cannot be found:
```
$ python3 -m mqtt_io mqtt-io.yaml
2025-02-23 21:13:57 mqtt_io.__main__ [ERROR] MqttIo crashed!
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 1450, in run
    self._init_sensor_modules()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 285, in _init_sensor_modules
    sens_config, "sensor", self.config["options"]["install_requirements"]
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 119, in _init_module
    "%s.%s.%s" % (MODULE_IMPORT_PATH, module_type, module_config["module"])
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'mqtt_io.modules.sensor.mhz19'
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/__main__.py", line 115, in <module>
    main()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 1450, in run
    self._init_sensor_modules()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 285, in _init_sensor_modules
    sens_config, "sensor", self.config["options"]["install_requirements"]
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 119, in _init_module
    "%s.%s.%s" % (MODULE_IMPORT_PATH, module_type, module_config["module"])
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'mqtt_io.modules.sensor.mhz19'
```

Also, it does not seem to be available:
```
$ pip3 install mhz19
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement mhz19 (from versions: none)
ERROR: No matching distribution found for mhz19
```

BenjiU, 2025-02-23:
Okay. you don't need to install a mhz19 package.  the mqtt-io-mhz19 Class relies on a direct serial connection.

But. You are using mqtt-io 2.3.0? from your post:
`Requirement already satisfied: mqtt-io in ./.local/lib/python3.7/site-packages (2.3.0)`

when i look in the history, mhz was added in version 2.4.2:
https://github.com/flyte/mqtt-io/releases/tag/2.4.2

That should be the reasen, why mqtt-io can't find mqtt-io-mhz19 module.

May you update your mqtt-io version?

mareklab, 2025-02-23:
Good point, I think you're right. 

I've successfully executed
```
pip3 install --upgrade git+https://github.com/flyte/mqtt-io.git@2.4.3
```

However, now I'm getting a different error 🤕:
```
$ python3 -m mqtt_io mqtt-io.yaml
2025-02-23 21:33:52 mqtt_io.__main__ [ERROR] MqttIo crashed!
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 1450, in run
    self._init_sensor_modules()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 285, in _init_sensor_modules
    sens_config, "sensor", self.config["options"]["install_requirements"]
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 131, in _init_module
    return module_class(module_config)
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/modules/sensor/__init__.py", line 22, in __init__
    self.setup_module()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/modules/sensor/mhz19.py", line 38, in setup_module
    self.ser = serial.Serial(
AttributeError: module 'serial' has no attribute 'Serial'
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/__main__.py", line 115, in <module>
    main()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/__main__.py", line 107, in main
    mqtt_gpio.run()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 1450, in run
    self._init_sensor_modules()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 285, in _init_sensor_modules
    sens_config, "sensor", self.config["options"]["install_requirements"]
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 131, in _init_module
    return module_class(module_config)
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/modules/sensor/__init__.py", line 22, in __init__
    self.setup_module()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/modules/sensor/mhz19.py", line 38, in setup_module
    self.ser = serial.Serial(
AttributeError: module 'serial' has no attribute 'Serial'
```

`pyserial` is already installed:
```
$ pip3 install pyserial
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pyserial in /usr/lib/python3/dist-packages (3.4)
```
Thanks for help so far... What I am missing? 
Unfortunately, I'm not too much of a Python developer...

mareklab, 2025-02-23:
I've found the solution. This MH-Z19 module is not compatible with Python 3.7, it started working fine after I've installed Python 3.8.

@BenjiU thanks for pointing me to a right direction!

closed 2025-02-23 by mareklab